MarlinFirmware/Marlin #26387: display bug

GPTKEY, 2023-11-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Chinese characters in the menu of the 12864 screen are displayed abnormally and become garbled, but the Chinese characters on the homepage are displayed normally.

### Bug Timeline

new bug

### Expected behavior

It feels like it's caused by python's character handling

### Version of Marlin Firmware

bugfix-2.1.x  

### Add-ons

board: skr1.1  LPC1768

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13241015/Marlin.zip)

GMagician, 2023-11-02:
Does your compiled firmware contains #26381? This should fix some utf8 chars, maybe it can solve your issue

ellensp, 2023-11-03:
Can you provide pictures of what your referring to?   I cant see an issue, but I can't read the language.

closed 2024-09-17 by github-actions[bot]