MarlinFirmware/Marlin #26388: Found CRITICAL Vulnerability in your website !

imchiragprajapati, 2023-11-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello I am Chirag Prajapati a Certified Penetration Tester and Ethical Hacker my License no is: 10188-161-078-1726.

We would like to inform you that, a security vulnerability affecting your website.

Affected Website:  marlin.crc.id.au

Following the ISO/IEC 29147 (“Information technology — Security techniques — Vulnerability disclosure”) guidelines, we verified the vulnerability's existence prior to notifying you. Please connect with us for technical details of the vulnerability, 
The researcher may also help remediate the vulnerability if you need any assistance.

For more information I request you to kindly revert me back so that I can share with you the report !
Looking forward to hearing from you !

### Version of Marlin Firmware

NA

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

The-EG, 2023-11-02:
Even if it this legit, marlin.crc.id.au is run by a community member and is not officially associated with Marlin or its official site.

imchiragprajapati, 2023-11-02:
Respected sir,
So whom I have to connect?
Can you guide me please?

*Looking forward to hearing from you! *

thisiskeithb, 2023-11-02:
> So whom I have to connect?

You’ll need to contact @CRCinAU since that is not our website.

closed 2023-11-02 by thisiskeithb

CRCinAU, 2023-11-03:
> Respected sir, So whom I have to connect? Can you guide me please? *Looking forward to hearing from you! *

You can either contact me via my website linked in my profile, or via a direct message on forums at https://marlin.crc.id.au/forum/