MarlinFirmware/Marlin #26389: [BUG] SSD1306 I2C display support appears to be missing

jonathanmlang, 2023-11-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I uncommented #define MKS_12864OLED_SSD1306 in configuration.h
Added pin definitions in the board file, pins_CREALITY_V4.h
#define I2C_SCL_PIN                           PB10
#define I2C_SDA_PIN                          PB11

Compiled and flahsed fine,
Blank display.

### Bug Timeline

Not sure

### Expected behavior

Expected the display to show the marlin interface

### Actual behavior

Nothing happens.

### Steps to Reproduce

I uncommented #define MKS_12864OLED_SSD1306 in configuration.h
Added pin definitions in the board file, pins_CREALITY_V4.h
#define I2C_SCL_PIN                           PB10
#define I2C_SDA_PIN                          PB11

### Version of Marlin Firmware

02010300

### Printer model

Creality ender 3 with custom display wired to exp3 cable

### Electronics

Creality v4.2.2 stm32 pcb

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13243699/Configuration.zip)

ellensp, 2023-11-02:
MKS_12864OLED_SSD1306 is not a I2C display, it is a SPI display

Your need to add appropriate SPI pins

ellensp, 2023-11-02:
Looking through the code I find there is

  #if ENABLED(ALTERNATIVE_LCD)
    #define U8G_CLASS U8GLIB_SSD1306_128X64_2X_I2C_2_WIRE       // 4 stripes
  #else

so you need to add #define ALTERNATIVE_LCD to your Configuration.h to use i2c mode

ellensp, 2023-11-02:
If your playing with a generic i2c SSD1306  module, you need #define U8GLIB_SSD1306

jonathanmlang, 2023-11-03:
Hello thanks for replying, Ive tried #define U8GLIB_SSD1306 but still no luck, no output at the pins as proven by an oscilloscope.

dbuezas, 2023-11-09:
Same here, I'm using an skr3 ez board on an Ultimaker 2.
I see the i2c clock for the i2c epprom on startup  but no clock at all if I disable it (not a single pulse).

I tried changing pins and software/hw i2c. It looks like Marlin isn't even trying to talk to the display at all.

ellensp, 2023-11-09:
having a play 

using a creality 4.2.2.controller and a 0.96" SSD1306 I2C module

I wired it up

```
  /**             LCD PORT
   *              ------
   *        PC6  | 1  2 | PB2
   *  (SDL) PB10 | 3  4 | PB11 (SDA)                               
   *        PB14   5  6 | PB13
   *        PB12 | 7  8 | PB15
   *         GND | 9 10 | 5V
   *              ------
   */

```

I added this code to Marlin/src/MarlinCore.cpp    after `SETUP_LOG("setup() completed.");`

```CPP
  SERIAL_ECHOLN("I2C scanner. Scanning ...");
  byte count = 0;
  Wire.setSDA(PB11);
  Wire.setSCL(PB10);
  Wire.begin();
  for (byte i = 1; i < 120; i++)
  {
    hal.watchdog_refresh();
    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
      {
      SERIAL_ECHOLN("Found address: ",i);
      count++;
      delay (1);
      }
  }
  SERIAL_ECHOLN ("Done.");
```

And `#include <Wire.h>`  after `#include <math.h>`

Marlin boots and says 

> I2C scanner. Scanning ...
> Found address: 60
> Done.

60 is 3C, the expected address of the LCD, so this confirms the address and the wiring is OK

ellensp, 2023-11-09:
Im getting clock and data...  cant say if its the correct data.  (this is well after the I2C scanner)

dbuezas, 2023-11-09:
Good idea with the i2c scan. I tried the same but with Software i2c and get this:

### Screen disconnected
```
10:18:16.948 > I2C scanner. Scanning ...
10:18:16.957 > Found address: 80
10:18:16.962 > Done.
```
That should be the onboard EEPROM

### Screen connected

```
10:18:16.948 > I2C scanner. Scanning ...
10:18:16.949 > Found address: 3
10:18:16.954 > Found address: 60
10:18:16.957 > Found address: 80
10:18:16.959 > Found address: 96
10:18:16.961 > Found address: 112
10:18:16.962 > Done.
```

**So my wiring should be correct.**

ellensp, 2023-11-09:
Odd, removing my I2C scanner code you dont get any output, as reported

adding  
```CPP
  Wire.setSDA(PB11);
  Wire.setSCL(PB10);
  Wire.begin();
```

above   SETUP_RUN(ui.init());  in Marlin/src/MarlinCore.cpp  and you get a display!!

[image]

This is far form a solution, more a hack, But my brain has quit for the day, will have to continue this later

ellensp, 2023-11-09:
I purpose this as a fix.   (basically it rearranges  marlinui.cpp to setup the i2c pins before setting up the display, and changes the pin variable type)

```DIFF
diff --git a/Marlin/src/lcd/marlinui.cpp b/Marlin/src/lcd/marlinui.cpp
index 463d406110..fa511b4882 100644
--- a/Marlin/src/lcd/marlinui.cpp
+++ b/Marlin/src/lcd/marlinui.cpp
@@ -218,6 +218,10 @@ constexpr uint8_t epps = ENCODER_PULSES_PER_STEP;
 
 void MarlinUI::init() {
 
+  #if HAS_U8GLIB_I2C_OLED && PINS_EXIST(I2C_SCL, I2C_SDA) && DISABLED(SOFT_I2C_EEPROM)
+    Wire.begin(uint8_t(I2C_SDA_PIN), uint8_t(I2C_SCL_PIN));
+  #endif
+
   init_lcd();
 
   #if HAS_DIGITAL_BUTTONS
@@ -274,10 +278,6 @@ void MarlinUI::init() {
     slow_buttons = 0;
   #endif
 
-  #if HAS_U8GLIB_I2C_OLED && PINS_EXIST(I2C_SCL, I2C_SDA) && DISABLED(SOFT_I2C_EEPROM)
-    Wire.begin(int(I2C_SDA_PIN), int(I2C_SCL_PIN));
-  #endif
-
   update_buttons();
 
   TERN_(HAS_ENCODER_ACTION, encoderDiff = 0);

```

ellensp, 2023-11-11:
Ive crated a PR to fix this, so closing this issue

closed 2023-11-11 by ellensp