MarlinFirmware/Marlin #26390: [BUG] False filament runouts with Filament motion sensor

mAd-DaWg, 2023-11-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

BTT Smart filament sensor v2 on ender 5 pro.

Sensor has an encoder that triggers every 2.88mm (according to [documentation](https://github.com/bigtreetech/smart-filament-detection-module/blob/master/V2.0/Manual/SFS%20V2.0%20User%20Manual%2020230210.pdf). Firmware is configured following documentation as well.

Default run out distance i configured is 3mm, but also tried 4mm and 6mm via menu at runtime. The sensor has an LED that flashes every time the encoder triggers. However marlin falsely flags filament runouts randomly, despite the encoder working correctly(with LED strobe as proof). 

This behaviour is not consistant as an earlier print experienced 0 false runout issues at all.

### Bug Timeline

old-ish, had similar experience with another filament encoder sensor

### Expected behavior

Marlin to not falsely trigger filament runouts

### Actual behavior

Marlin randomly triggers filament runouts

### Steps to Reproduce

1. Setup and configure a filament runout sensor, using a runout distance close to the sensors rating
2. Run a print
3. Experience a false runout
4. Increment the runout distance
5. Still experience random false runouts

### Version of Marlin Firmware

latest marlin bugfix 2.1xx (2 November 2023)

### Printer model

Ender 5 Pro

### Electronics

Stock electronics

### Add-ons

BTT Smart filament sensor v2, blTouch sensor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13245624/Configuration.zip)
[CE5_speedDrive_Ender5_Remix_10mm.zip](https://github.com/MarlinFirmware/Marlin/files/13245625/CE5_speedDrive_Ender5_Remix_10mm.zip) <- print that had filament runout immediately after printing initial purge line despite correct sensor strobing

ellensp, 2023-11-03:
Not current bugfig config files

Is missing the section

```
    #if ENABLED(FILAMENT_MOTION_SENSOR)
      //#define FILAMENT_SWITCH_AND_MOTION
      #if ENABLED(FILAMENT_SWITCH_AND_MOTION)
        #define NUM_MOTION_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_MOTION#_PIN for each.
        //#define FIL_MOTION1_PIN    -1

        // Override individually if the motion sensors vary
        //#define FIL_MOTION1_STATE LOW
        //#define FIL_MOTION1_PULLUP
        //#define FIL_MOTION1_PULLDOWN

``` 

No FIL_MOTION1_PIN defined

mAd-DaWg, 2023-11-03:
It definitely is the latest because i downloaded and compiled the firmware yesterday from the bug-fix branch, i must have accidentally taken the section out when comparing configurations.. Even if the above section is missing, it changes nothing as i am not using filament switch and motion, i am only using motion.

mAd-DaWg, 2023-11-03:
Im not sure what happened, but that config zip is wrong. Here is the correct one
[Configuration_new.zip](https://github.com/MarlinFirmware/Marlin/files/13245885/Configuration_new.zip)