MarlinFirmware/Marlin #26395: [FR] XY compensation for Delta printers

kocickamia, 2023-11-05:
### Describe the feature you want

Hello.
If my cartesian printer prints object for example 100x100mm square, and result is 100x99.5 rectangle, then i may change steps/mm for appropriate motor. But this method is not applicable for Delta printers, as they needs the same steps/mm for all three motors.
In Configuration.h is item SKEW_CORRECTION where i may compensate rhombus to square. Could be add item for compensate rectangle to square in case of a Delta printer?

EvilGremlin, 2023-11-05:
Fix the issue, not the symptoms. SKEW_CORRECTION is already extremely dubious feature, personally i would never merge it. If you have this result then your geometry values are wrong, or you have something floppy in mechanics.

RatmanL, 2023-11-05:
Personally, I think that if this feature exists and is used in other types of printers, it should be applicable to the Delta as well. 
After all, a mechanical solution to a problem is not always easy to measure and fix. That is the reality. And if there is an option to treat this in software I would definitely take advantage of it.

EvilGremlin, 2023-11-05:
No, XY compensation is completely diffrerent ftom skew correction, this is whole new feature. I'm of the opinion that such software compensation is only harmful, becasue it panders to bad hardware manufacturers and bad user practices.

RatmanL, 2023-11-05:
I respect your opinion, but I still think that there should be a way to compensate for this deviation in software. If it is a big difference I agree completely with your statement.

jmz52, 2023-11-24:
> But this method is not applicable for Delta printers, as they needs the same steps/mm for all three motors.

They don't. You can use different values for each carriage.
Personally, I strongly recommend to calibrate steps/mm for each carriage before doing delta calibration.