MarlinFirmware/Marlin #26396: [FR] MKS THR36 and MKS THR42 CAN bus (RP2040 Pico based) remote tool head support

rondlh, 2023-11-05:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

Yes,  MKS THR36 and MKS THR42 CAN bus (RP2040 Pico based) tool head support

### Describe the feature you want

Support for Marlin on MKS THR36 and MKS THR42.
https://github.com/makerbase-mks/MKS-THR36-THR42-UTC

I tried this code:
https://github.com/thinkyhead/Marlin/tree/bf2_wip_rp2040_skr_pico_PR

Which already makes most things work, but I cannot get the TMC2209 to work in serial configuration mode, if I set the driver to TMC2209 then things go wrong, even the host communication breaks down.
If Marlin can run on this board, then I will work on implementing the CAN communication allowing for a remote tool head in Marlin over CAN.

[image]

thisiskeithb, 2023-11-05:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/7735

closed 2023-11-05 by thisiskeithb

rondlh, 2023-11-05:
> Duplicate of #7735

Should I delete it?