MarlinFirmware/Marlin #26398: [BUG] (Speed Update delayed by 1 line)

rflulling, 2023-11-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

9c3704c 2023-08-11 Bug Fix
https://github.com/MarlinFirmware/Configurations/commit/9c3704c90679fe4d349450c652e8c631ad6d748b

The very first line is either accepting the update, or it is executing the prior speed set by homing which is 2.1mm/s.
The expected behavior is that the movement on the line, is instant. 

I had seen this issue before but thought it was just a issue in the way i was writing gcode. So I had added an extra line of nonsense movement data to force the update of the speed. Chasing other issues, in the hardware, I didn't stop to pay attention to the way the code was being handled at the time. But with the hardware issues resolved. Its time to re calibrate. For whatever reason, the manual leveling isn't responding on the LCD. So I used S3D 4.11, which still has leveling built in. I will recompile later, and hopefully it fixes on screen leveling.

When running leveling in S3D it sends very simple commands as Absolute coordinates. 
G1 X40.000 Y40.000 Z10.000 F126  // Moves to X40 Y40 Z10 at 2.1mm 
G1 X260.000 Y610.000 Z10.000 F4800 // Moves to X260 Y610.00 F4800
G1 X260.000 Y610.000 Z0.000 F126 // Z0 fails, to fast

The very first line is either accepting the update, or it is executing the prior speed set by homing which is 2.1mm/s.

### Bug Timeline

N/A

### Expected behavior

G1 X200 F1000 // X moves to 200 at 16.667mm/s
G1 Z0 F126 // Z moves to 0 at 2.1mm/s
G1 Z10 F126 // Z moves to 10 at 2.1mm/s
G1 X500 F1000 // X moves to 500 at 16.667mm/s
G1 Y500 F1000 // Y moves to 500 at 16.667mm/s

The expected behavior is that the movement on the line, is instant. 

### Actual behavior

G1 X200 F1000 // X moves to 200 at 16.667mm/s
G1 Z0 F126 // Z moves to 0 at 16.667mm/s
G1 Z10 F126 // Z moves to 10 at 2.1mm/s
G1 X500 F1000 // X moves to 500 at 2.1mm/s
G1 Y500 F1000 // Y moves to 500 at 16.mm/s

### Steps to Reproduce

G1 X40.000 Y40.000 Z10.000 F126 
G1 X260.000 Y610.000 Z10.000 F4800
G1 X260.000 Y610.000 Z0.000 F126

If one is to send a series of commands, slowly, by hand, and the speed difference between them is significant, it should be pssoible to directly observe and confirm this behavior. 

### Version of Marlin Firmware

2.1.x Bug Fix

### Printer model

Custom

### Electronics

SKR PRO v1.1

### LCD/Controller

BIGTREETECH TFT35-E3 V3.0.1

### Other add-ons

TMC5160

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Simplify3D

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Firmware has not been altered in anyway. 

9c3704c 2023-08-11 Bug Fix
https://github.com/MarlinFirmware/Configurations/commit/9c3704c90679fe4d349450c652e8c631ad6d748b

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13261309/Configuration.zip)

thisiskeithb, 2023-11-06:
> 9c3704c 2023-08-11 Bug Fix

This isn’t a valid `bugfix-2.1.x` commit, but you’re pretty far behind current bugfix based on the provided date.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

rflulling, 2023-11-06:
Installing:
0bdbf52 2023-11-05 Bug Fix
https://github.com/MarlinFirmware/Marlin/commit/0bdbf52bc61f054b418b623bb18256edb4a2dd13

rflulling, 2023-11-07:
As feared, new issues were imported. So having to work around since it does effect the test. 
With a 30:1 reduction installed. Already had a limit of F156, but after this latest update new limit is F72. I don't know what all changed but, had to find and reduce homing, and max rates for everything that effect that axis just retest the bug, otherwise I would wake up the entire block with the motor crying. 

With everything limited to 1x it seems to be working. I mean the motor is correctly spinning away at high speed. But why it has to go slower I cannot identify. 

The only reason the bug was identified was that in the transition between lines, since Z and XY have different speeds the commands are sent separate. However since Z cannot go F1800 and is limited to F72, the motor screams and nothing moves. This helped me to identify that there was an issue, that the Z line was still executing at the XY speed, even though it was already set on that line. 

As of the update. I had to tweak so much, it is possible I am preventing the bug from being an issue. I think I need to allow the Z to go faster than it should in order for it to be tell me if it is being told to go too fast. It would be easy to say that the bug is not in this version of BF. But I have not yet confirmed that. Still testing.

closed 2024-03-18 by github-actions[bot]