MarlinFirmware/Marlin #26402: POWER LOSS RECOVERY

louro67, 2023-11-08:
### Is your feature request related to a problem? Please describe.

When this function is activated, the printer does everything but does not go to the right layer of the print. Is there any way to calibrate this?

### Are you looking for hardware support?

SKR 1.3

### Describe the feature you want

When this function is activated, the printer does everything but does not go to the right layer of the print. Is there any way to calibrate this?

### Additional context

When this function is activated, the printer does everything but does not go to the right layer of the print. Is there any way to calibrate this?

ellensp, 2023-11-08:
This is not a feature request or a bug report

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-11-08 by ellensp

thinkyhead, 2024-01-10:
We'll have #26365 merged soon, and that should fix all issues with position in power-loss recovery.