MarlinFirmware/Marlin #26408: [BUG] Missing letters in platform.io

Rastafaraaj, 2023-11-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I think there is something wrong with platform.io, marlin auto build doesn't see nothing, File has only 4kB.

### Version of Marlin Firmware

latest bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-11-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

Rastafaraaj, 2023-11-10:
Im really sorry - my bad. There is no problm with platform.io, someone just removed "MOTHERBOARD"  #define MOTHERBOARD BOARD_BTT_SKR_V1_3, that's why auto build didn't saw nothing.

closed 2023-11-11 by thisiskeithb