MarlinFirmware/Marlin #26417: [BUG] Latest bugfix won't compile due to excessive Z_Probe_Zmin and Zmax values

rq3, 2023-11-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The latest bugfix won't successfully compile due to a minor error in configuration.h:

#define PROBE_OFFSET_ZMIN   -30   // (mm)
#define PROBE_OFFSET_ZMAX    30   // (mm)

default values are in excess of 20, so an error is thrown.

### Bug Timeline

new

### Expected behavior

I expected a clean compile.

### Actual behavior

Compile throws an error due to new, excessively large z probe offset values.

### Steps to Reproduce

1) Compile latest Marlin bugfix under Vscode.
2) Observe error.

### Version of Marlin Firmware

latest bugfix November 11, 2023

### Printer model

Anycubic Predator

### Electronics

stock, Trigorilla Pro with STM32F103

### LCD/Controller

stock

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13326243/Configuration.zip)

ellensp, 2023-11-12:
In Marlin/src/inc/SanityCheck.h 

  PROBE_OFFSET_ASSERT("PROBE_OFFSET_ZMIN", PROBE_OFFSET_ZMIN, -20, 20);
  PROBE_OFFSET_ASSERT("PROBE_OFFSET_ZMAX", PROBE_OFFSET_ZMAX, -20, 20);

Added in https://github.com/MarlinFirmware/Marlin/pull/26267

Will trigger on all these example Configs 

config/examples/delta/FLSUN/QQS-Pro/Configuration.h:#define PROBE_OFFSET_ZMIN   -25   // (mm)
config/examples/delta/FLSUN/Q5-nano_v2/Configuration.h:#define PROBE_OFFSET_ZMIN   -40   // (mm)
config/examples/delta/FLSUN/Q5-nano_v1/Configuration.h:#define PROBE_OFFSET_ZMIN   -40   // (mm)
config/examples/delta/Anycubic/Predator/Configuration.h:#define PROBE_OFFSET_ZMIN   -30   // (mm)
config/examples/delta/Anycubic/Predator/Configuration.h:#define PROBE_OFFSET_ZMAX    30   // (mm)
config/examples/delta/Anycubic/Kossel/Configuration.h:#define PROBE_OFFSET_ZMIN   -40   // (mm)
config/examples/delta/Anycubic/Kossel Linear Plus/Configuration.h:#define PROBE_OFFSET_ZMIN   -40   // (mm)
config/examples/AnyCubic/Chiron/Configuration.h:#define PROBE_OFFSET_ZMIN   -30   // (mm)

CRCinAU, 2024-01-31:
For what its worth, this same problem affects the X/Y offsets too.

Multiple users require offsets greater than 50mm, and setting higher will cause a build failure.

ellensp, 2024-01-31:
yes these PROBE_OFFSET min and max setting are just a PITA from a user experience point of view

In my view the defaults should be enabled and set to

#define PROBE_OFFSET_XMIN -X_BED_SIZE   // (mm)
#define PROBE_OFFSET_XMAX  X_BED_SIZE   // (mm)
#define PROBE_OFFSET_YMIN -Y_BED_SIZE   // (mm)
#define PROBE_OFFSET_YMAX  Y_BED_SIZE   // (mm)
#define PROBE_OFFSET_ZMIN -Z_MAX_POS   // (mm)
#define PROBE_OFFSET_ZMAX  Z_MAX_POS   // (mm)

Then if you wish to tie your hands up with reducing them, you can, but it is not the default.

NB this wont work currently, as bed sizes are defined after the PROBE_OFFSETS, so a reordering of the config file would be needed

CRCinAU, 2024-01-31:
Taggings @thinkyhead for awareness / input.

ellensp, 2024-01-31:
actually they are in Conditionals_post.h if not set in Configs, so setting the defaults to be full size is easy

ellensp, 2024-01-31:
ie 

```
#if ANY(MESH_BED_LEVELING, HAS_BED_PROBE)
  #ifndef PROBE_OFFSET_ZMIN
    #define PROBE_OFFSET_ZMIN -Z_MAX_POS
  #endif
  #ifndef PROBE_OFFSET_ZMAX
    #define PROBE_OFFSET_ZMAX  Z_MAX_POS
  #endif
#endif
#if HAS_BED_PROBE
  #ifndef PROBE_OFFSET_XMIN
    #define PROBE_OFFSET_XMIN -X_BED_SIZE
  #endif
  #ifndef PROBE_OFFSET_XMAX
    #define PROBE_OFFSET_XMAX  X_BED_SIZE
  #endif
  #ifndef PROBE_OFFSET_YMIN
    #define PROBE_OFFSET_YMIN -Y_BED_SIZE
  #endif
  #ifndef PROBE_OFFSET_YMAX
    #define PROBE_OFFSET_YMAX  Y_BED_SIZE
  #endif
  
```