MarlinFirmware/Marlin #26418: [BUG] Marlin bugfix-2.1.2.1 has blank display, and continuously re-boots

rq3, 2023-11-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin Release 2.1.2.1 of 2023-5-16 works fine.
Marlin Patched 2.1.2.1 of 2023-5-16 works fine.
Marlin bugfix-2.1.2.1 yields a blank display and continuous re-boot behavior.

### Bug Timeline

First started with bugfix of May 17, 2023. All bugfixes since then have shown this behavior.

### Expected behavior

I expect the latest bugfix to function.

### Actual behavior

The Release 2.1.2.1 bugfixes since May 17, 2023 show a blank screen, and re-boot continuously approximately every two seconds.

### Steps to Reproduce

1) Compile any 2.1.2.1 bugfix after May 17, 2023.
2) Install the firmware on the printer.
3) Turn the printer on, and observe a blank display with the controller re-booting continuously.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Anycubic Predator

### Electronics

stock. STM32F103 and 3.5 inch 480x320 TFT display

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

In all cases, fresh firmware.bin files were compiled under Platformio in VSCode using the Github Marlin releases and their associated configuration files (attached).

In all cases, the ONLY changes made to the associated configuration files were:
1) In configuration.h, the "#define MESH_EDIT_GFX_OVERLAY" was disabled as it is not appropriate for this display, and should be disabled by default anyway.
2) In configuration.h, the display was changed to Color_UI.

[image]
[image]
[image]
[Marlin_Current.zip](https://github.com/MarlinFirmware/Marlin/files/13326386/Marlin_Current.zip)
[Marlin_Patched.zip](https://github.com/MarlinFirmware/Marlin/files/13326388/Marlin_Patched.zip)
[bugfix-2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/13326390/bugfix-2.1.x.zip)

ellensp, 2023-11-12:
you have 3 images, and no description of what they related to. 1 non working two working..

ellensp, 2023-11-12:
There also seems to be multiple TRIGORILLA PRO Boards... the weird one with a foot? https://github.com/AnyCubic-i3-Mega-Model-4/ReverseEngineerTrigorillaPro  or a newer A or B square  board version?
What exact controller do you have?

ellensp, 2023-11-12:
continuation of https://github.com/MarlinFirmware/Marlin/issues/25852

rq3, 2023-11-12:
@ellensp, the blank display and continuous re-boot is what you get with any bugfix after May 17, 2023. As I mentioned, the released version of 2.1.2.1 and the patched version of 2.1.2.1 work fine. It is only the bugfixes on and after May 17, 2023 that have issues. I determined this with a binary search of bugfixes, and have been trying almost every bugfix since May 17.

My Trigorilla Pro is the "weird one with a foot", and uses an STM32F103ZET6 CPU running at 72 MHz. The Predator never used any other version of board or CPU.

Yes, this is a continuation of an ongoing issue that has not been addressed, and needs to be re-examined, as #25852 is tagged as stale, and closing soon. I'll help in any way that I can.

thisiskeithb, 2023-11-12:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/25852

closed 2023-11-12 by thisiskeithb

thisiskeithb, 2023-11-12:
> Yes, this is a continuation of an ongoing issue that has not been addressed, and needs to be re-examined, as https://github.com/MarlinFirmware/Marlin/issues/25852 is tagged as stale, and closing soon.

You need to do what the stale bot comment says or an issue will be closed due to inactivity.

> This issue has had no activity in the last 60 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 10 days.