MarlinFirmware/Marlin #26420: [BUG] "🔨 Specific package versions (#26265)" breaks compiling on SKR 3

thisiskeithb, 2023-11-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

"🔨 Specific package versions (#26265)" breaks compiling on SKR 3

### Bug Timeline

Started with d159ec5 / #26265

### Expected behavior

Marlin to compile

### Actual behavior

Marlin does not compile:

<details><summary>details:</summary>
<p>

```prolog
TypeError: argument of type 'NoneType' is not iterable:
  File "/opt/hostedtoolcache/Python/3.7.17/x64/lib/python3.7/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/buildroot/share/PlatformIO/scripts/generic_create_variant.py", line 35:
    if '@' in uri:
========================= [FAILED] Took 47.90 seconds =========================

Environment      Status    Duration
---------------  --------  ------------
STM32H743VI_btt  FAILED    00:00:47.900
==================== 1 failed, 0 succeeded in 00:00:47.900 ====================
```

</p>
</details>

### Steps to Reproduce

1. Download current `bugfix-2.1.x` / anything after d159ec5 & add configs from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5%20Plus/BigTreeTech%20SKR%203
2. Compile
3. See error

### Version of Marlin Firmware

`bugfix-2.1.x` / 235ad4d

### Printer model

Any SKR 3 config

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Use SKR 3 config from Configurations repo: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5%20Plus/BigTreeTech%20SKR%203

ellensp, 2023-11-13:
This fixes it

```DIFF
diff --git a/buildroot/share/PlatformIO/scripts/generic_create_variant.py b/buildroot/share/PlatformIO/scripts/generic_create_variant.py
index 457ef95d5a..f097121d99 100644
--- a/buildroot/share/PlatformIO/scripts/generic_create_variant.py
+++ b/buildroot/share/PlatformIO/scripts/generic_create_variant.py
@@ -32,6 +32,8 @@ if pioutil.is_pio_build():
         platform_name = framewords[platform.__class__.__name__]
     else:
         uri = PackageSpec(platform_packages[0]).uri
+        if uri is None:
+            uri = [] # 👈️ set to empty list if None
         if '@' in uri:
             platform_name = re.sub(r'@.+', '' , uri)
         else:

```

thisiskeithb, 2023-11-13:
> This fixes it

Thanks! Sure does.