MarlinFirmware/Marlin #26421: [BUG] Marlin 2.1.x won't build with BTT_TFT35_SPI_V1_0 after using BigTreeTech repository instructions

RoscoeTheDog, 2023-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a **BTT SKR 1.4 Turbo** motherboard and a **TFT3.5 v3.0 (non-E variant)** touch screen display. I am trying to build Marlin 2.1.x (latest) branch for production. There are many required options needed according to the main bigtreetech repository found [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware). I've followed these instructions, uncommented the `BTT_TFT35_SPI_V1_0` and I get the following compiler errors:

tft_spi.cpp
```
'TFT_SPI_DEVICE' was not declared in this scope
'LPC_SSPx' was not declared in this scope; did you mean 'LPC_SSP0'?
```

tft_spi.h
```
#error "Invalid TFT SPI configuration."
'LPC_SSPx' was not declared in this scope; did you mean 'LPC_SSP0'?
```

I've double checked that I did not skip over any options that are required as per instructions, and prior to trying to setup the LCD, I could build the firmware without all the display and required options enabled. Please see [this link](https://drive.google.com/file/d/1RW_1GdJeFB_eoviBNMQ72TFBBQVDxhet/view?usp=sharing) for archive of firmware configuration. its on gdrive because github won't allow <25mb.

Kind regards

### Version of Marlin Firmware

2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-11-13:
BigTreetech's TFT35 V3.0 is not a TFT35 SPI V1.0. They are standalone computer's running their own firmware / G-code senders that talk via serial and emulate `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` / `CR10_STOCKDISPLAY` via the two EXP cables.

Disable `BTT_TFT35_SPI_V1_0` and follow the rest of the instructions in BTT's touchscreen repo under the [Marlin Dependencies section](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies) & [Marlin Mode Setup](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-mode-setup) where you'll enable `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, among many other settings instead.

closed 2023-11-13 by thisiskeithb

ellensp, 2023-11-13:
BTT TFT35 SPI V1.0
[image]

BTT TFT35 V3.0
[image]

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

RoscoeTheDog, 2023-11-13:
Thank you for the quick response. I did not see this step as it was in the "marlin-mode" segment of the table of contents, which seems vaguely labeled for people who don't know what marlin emulation is.

Regarding support, I understand it takes time to respond and these are not priorities on the team's radar. I will refrain from posting further and take to discord it becomes more complex.

I implore you to take one more look for me please. I enabled `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` and disabled the other, which lead to other compiler errors. It seems there is something wrong with the I2C dependencies?

```
Building in release mode
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o: No such file or directory
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o] Error 1
```

Kind regards

thisiskeithb, 2023-11-13:
> I implore you to take one more look for me please. I enabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and disabled the other, which lead to other compiler errors. It seems there is something wrong with the I2C dependencies? The error seems unclear and googling around has been less fruitful.

After changing the display type from `BTT_TFT35_SPI_V1_0` to `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, your config compiles fine:

```prolog
Linking .pio/build/LPC1769/firmware.elf
Building .pio/build/LPC1769/firmware.bin
Checking size .pio/build/LPC1769/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  43.7% (used 14312 bytes from 32736 bytes)
Flash: [====      ]  42.6% (used 202332 bytes from 475136 bytes)
============================================== [SUCCESS] Took 38.80 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        SUCCESS   00:00:38.796
============================================== 1 succeeded in 00:00:38.796 ==============================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

[26421 - corrected configs.zip](https://github.com/MarlinFirmware/Marlin/files/13335020/26421.-.corrected.configs.zip)

As ellensp stated above, please use the linked resources for further troubleshooting.

RoscoeTheDog, 2023-11-13:
> After changing the display type from `BTT_TFT35_SPI_V1_0` to `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, your config compiles fine:
>
> ```prolog
> Linking .pio/build/LPC1769/firmware.elf
> Building [...]

Odd! I tried again this morning and now it compiled. I think VSCode and platformio got closed down at one point and I had to reinit the IDE. It's possible platformio did not configure itself all the way properly when I reopened the project.

Thank you again. You both have been great help 🙏