MarlinFirmware/Marlin #26422: [BUG] Using up and down buttons does not work

jonathanmlang, 2023-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I defined BTN_UP and BTN_DOWN in pin_<board>.h
I may have done it wrong but there is no documentation covering this.

### Bug Timeline

Not sure

### Expected behavior

Expected pressing the up and down buttons I assigned, to navigate the menu.

### Actual behavior

Nothing happens

### Steps to Reproduce

I defined BTN_UP and BTN_DOWN in pins_<board>.h or in my case the pins_CREALITY_V4.h

### Version of Marlin Firmware

02010201

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration files are unmodified from stock creality ender 3 board 4.2.2 in configurations repository

thisiskeithb, 2023-11-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-11-13 by thisiskeithb

jonathanmlang, 2023-11-13:
These suggestions did not work.
Marlin documentation does not specify a solution.
Marlin forum returned no useful results

Therefore I can only assume this functionality isn't available to ive changed it to a Feature Request.

thisiskeithb, 2023-11-13:
> Therefore I can only assume this functionality isn't available

Buttons already work. Our Discord server is probably the best place to receive technical support, so please join!

jonathanmlang, 2023-11-14:
I was ignored on the discord server.
Im still having difficulty with getting the up/down buttons to work. Ive added BTN_UP and BTN_DOWN to the pins file and its still not working. Went into the buttons.h file and changed SET_INPUT(BTN_UP); to SET_INPUT_PULLUP(BTN_UP);, declared HAS_ENCODER_ACTION in the pins file to enable the button related function at line 1326 of marlinui.cpp. Now I get a pullup at the button pins but they still do nothing.
This indicates to me this is actually a BUG and this issue needs reopening. Thanks

jonathanmlang, 2023-11-18:
Ok I have solved the problem. the issue was:
Line 525 in Conditionals_LCD.h

`#if IS_ULTIPANEL && DISABLED(NO_LCD_MENUS)`

needs changing to:

`#if ((IS_ULTIPANEL || IS_NEWPANEL) && DISABLED(NO_LCD_MENUS))`