MarlinFirmware/Marlin #26424: [BUG] Geeetech A10M LCD Menus Scrambled

SparkyDan555, 2023-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Some LCD menus are misaligned after scrolling to the bottom. See below photos. 

Filament Change Menu
[image]

Temperature Configuration Menu (notice random characters next to Ambient Co.)
[image]

Bottom of Temperature Configuration Menu. 50 is from Autotune Heated Bed option.
[image]

This is similar to an issue I previously reported #26148 

Only some menus are affected, others work fine so I have ruled out any interference in the electronics.

### Bug Timeline

First noticed in bug fix approx 2-3 weeks ago.

### Expected behavior

LCD Menus properly aligned and formatted

### Actual behavior

When scrolling towards the bottom of LCD menus, the menu structure becomes misaligned and generally scrambled. 

### Steps to Reproduce

1. Flash latest bug fix
2. Navigate to affected LCD menu page
3. Scroll to bottom

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Nov 13 2023 20:12:25)

### Printer model

Geeetech A10M

### Electronics

Stock GT2560 V1.4

### LCD/Controller

Stock Character LCD

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/13341927/Archive.zip)

ellensp, 2023-11-14:
User seems a little confused
The Temperature Menu shown is Configuration | Advanced Settings | Temperature

SparkyDan555, 2023-11-14:
> User seems a little confused
> 
> The Temperature Menu shown is Configuration | Advanced Settings | Temperature
> 
> 
> 
> 
> 
> 
> 
> 

Yes, the "temperature configuration" menu in Configuration > Advanced Settings > Temperature as opposed to the Temperature page on the main menu. No confusion here.

ellensp, 2023-11-14:
Tracking the last line issue in Advanced Temperature... 

What ive found so far

In Marlin/src/lcd/HD44780/marlinui_HD44780.cpp I added some serial echos to see what was going on

```DIFF
diff --git a/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp b/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
index 2564d4f96e..45b55e12f9 100644
--- a/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
+++ b/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
@@ -1223,9 +1223,12 @@ void MarlinUI::draw_status_screen() {
   // Draw a menu item with a (potentially) editable value
   void MenuEditItemBase::draw(const bool sel, const uint8_t row, FSTR_P const ftpl, const char * const inStr, const bool pgm) {
     const uint8_t vlen = inStr ? (pgm ? utf8_strlen_P(inStr) : utf8_strlen(inStr)) : 0;
+    SERIAL_ECHOLN("vlen:",vlen);
     lcd_put_lchar(0, row, sel ? LCD_STR_ARROW_RIGHT[0] : ' ');
     uint8_t n = LCD_WIDTH - 2 - vlen;
+    SERIAL_ECHOLN("n(1):",n);
     n -= lcd_put_u8str(ftpl, itemIndex, itemStringC, itemStringF, n);
+    SERIAL_ECHOLN("n(2):",n);
     if (vlen) {
       lcd_put_u8str(F(":"));
       for (; n; --n) lcd_put_u8str(F(" "));
```

On displaying the last line on the menu you get
vlen: 3
n(1): 15
n(2): 255

lcd_put_u8str is returning 255, which results in 255 spaces being incorrectly added.

Just noting this down, since I dont know when ill get back to this...

ellensp, 2023-11-14:
Partial fix

```DIFF
diff --git a/Marlin/src/lcd/lcdprint.cpp b/Marlin/src/lcd/lcdprint.cpp
index c84a695c78..aeaf9c27cc 100644
--- a/Marlin/src/lcd/lcdprint.cpp
+++ b/Marlin/src/lcd/lcdprint.cpp
@@ -116,8 +116,8 @@ lcd_uint_t expand_u8str_P(char * const outstr, PGM_P const ptpl, const int8_t in
  */
 lcd_uint_t lcd_put_u8str_P(PGM_P const ptpl, const int8_t ind, const char *cstr/*=nullptr*/, FSTR_P const fstr/*=nullptr*/, const lcd_uint_t maxlen/*=LCD_WIDTH*/) {
   char estr[maxlen + 2];
-  const lcd_uint_t outlen = expand_u8str_P(estr, ptpl, ind, cstr, fstr, maxlen);
-  lcd_put_u8str_max(estr, maxlen * (MENU_FONT_WIDTH));
+  expand_u8str_P(estr, ptpl, ind, cstr, fstr, maxlen);
+  const lcd_uint_t outlen = lcd_put_u8str_max(estr, maxlen * (MENU_FONT_WIDTH));
   return outlen;
 }
 ```

[image]

but when you go into the edit screen

[image]

thinkyhead, 2023-11-23:
Thanks for the report! This is now patched in `bugfix-2.1.x`.

SparkyDan555, 2023-11-23:
> Thanks for the report! This is now patched in `bugfix-2.1.x`.

I just did a quick build of bugfix-2.1.x and checked this commit is included and the issue still exists in the steps/mm menu. I will test some more later.

ellensp, 2023-11-23:
Steps/mm menu looks ok, but the value editing screen is down one line, is that what your referring to?

SparkyDan555, 2023-11-23:
Hopefully a video might demonstrate this better: https://drive.google.com/file/d/1cFCz0oyNODbwqhyPhQ28HUI1FluS-dBT/view

I do also experience the value on the editing screens being one line down, but that is not the main point I'm trying to raise. The problem tends to exist when you scroll right to the bottom of the menu, the last menu item disappears, showing only the value at some random position on the screen. When you scroll back up, it becomes generally scrambled/misaligned and doesn't navigate as it should.

commit ded942a4 (2023-11-23): 🐛 Fix expand_u8str_P string substitution

Fixes #26424
closed 2026-10-08 by thinkyhead