MarlinFirmware/Marlin #26425: [BUG] Latest Bugfix gcode M190 causes echo:busy: paused for user

qthegaijin, 2023-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Every instance of M190 causes this message:
Recv: echo:Press button to resume
Recv: echo:busy: paused for user

In Octoprint Terminal with no way to press a button or resume (Latest Marlin Firmware does not work with the LongerLK5Pro Screen so I do not have it attached). I have tried using M108 or similar in the Gcode but nothing happens. Code M140 works without issues to heat the bed to the desired temp.

### Bug Timeline

New-ish, started a few days ago when updating firmware

### Expected behavior

Expected: M190 S to heat the bed to the set degree temp then proceed with the rest of the code.

### Actual behavior

M190 S heats the bed to the desired temp then gets stuck in a loop sending the terminal message:

Recv: echo:Press button to resume
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user

With no way to exit the loop. Happens with all Gcode with M190 S listed.

### Steps to Reproduce

Tested with the bedlevelvisualizer Gcode:

M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M420 S0  ; Turning off bed leveling while probing, if firmware is set
         ; to restore after G28
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display

Bed leveling wont get past M190 S60.

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Longer LK5 Pro

### Electronics

Octoprint Raspi

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[M190 Longer Error Configs.zip](https://github.com/MarlinFirmware/Marlin/files/13342914/M190.Longer.Error.Configs.zip)

qthegaijin, 2023-11-13:
Just to add to this, it seems Hot End Wait code M109 also causes the exact same bug.

thisiskeithb, 2023-11-14:
You should enable `HOST_ACTION_COMMANDS` & `HOST_PROMPT_SUPPORT`.

classicrocker883, 2023-11-14:
**M108** only works `#if DISABLED(EMERGENCY_PARSER)`

closed 2024-03-18 by github-actions[bot]