MarlinFirmware/Marlin #26431: Independent axis movements

Developer310301, 2023-11-16:
### Describe the feature you want

I need that each axis can be run independently from others like:
- G0 X100 //move the x-axis
- G0 Y100 //move the y-axis while x-axis is running
and so on, but i haven't found any firmware configuration that can perform this.

ellensp, 2023-11-16:
This is not possible.   

Marlin is designed that moves involving multiple axis are synchronized, Otherwise gcode is processed sequentially 
These are design requirements.

Otherwise the moves are rather non deterministic. 

Eg your example.
The X would start moving first, then the Y moves will start at a random X position. X move would finish and Y would still be moving

You would get a line like the following piece of primitive ascii art

```
¯\
  |

```

This is not a feature it is a complete change in design philosophy for something other than a 3d printer.

thisiskeithb, 2023-11-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24474

closed 2023-11-16 by thisiskeithb