MarlinFirmware/Marlin #26432: [FR] Add separate GCODE commands for turning Input Shaping ON/OFF, without changing frequencies

Tupson444, 2023-11-16:
### Is your feature request related to a problem? Please describe.

If I have a file sliced for Input Shaping, and the file has commands to gradually reduce frequency on the Y axis (since it is a bedslinger and added weight changes the frequency, this can be set in PrusaSlicer), and mid print I realize something is wrong with my Input Shaping settings (for example), I could try to turn it OFF (and reduce acceleration), but I would not be able to because it will be turned ON as soon as the next M593 command is executed.

### Describe the feature you want

I think it would be useful to have separate commands for changing Input Shaping state (on/off), different from the commands for changing frequencies. Currently command `M593 X F35` will set X frequency to 35 AND turn IS on, while `M593 X F0` will both turn it off and "erase" frequency settings.

New commands could be something like `M593 X S1` and `M593 X S0` (or whatever else), while `M593 X F35` should for example only change the frequency and NOT turn IS on (similar as `M207` sets firmware retraction parameters, while `M209` turns it on/off).

### Additional context

I had opened a similar issue before (#25814), but it was about having IS turned off by default, and later I realized there are other reasons for having separate commands, so I'm opening this issue now.

Another use case example: I slice several files, some to be printed with IS and some not. However, in the meantime I might remeasure frequencies on the machine and change them through the menu. In this case it would be good to be able to have command to turn IS on (e.g. `M593 X S1`) in the file that I want to be printed with IS. Currently I would need to reslice all the files intended to be printed with IS whenever I change something on the machine and measure new frequencies. And I'm sure there are more examples.

I know this would complicate things for people who would expect the commands to continue working the same way as they are now, but if the change is to be made, it's better earlier than later.

narno2202, 2023-12-20:
@Tupson444 , if you look at the way IS is implemented you will see that is not intended to be switched on or off on the fly. This could be done with FT_MOTION on the fly (`M490 S1` : `FT_MOTION` is enabled without shaping function) but it's not a good idea to disable the shaper during an ongoing print.

Tupson444, 2023-12-20:
I understand, this may be irrelevant for that case then. But for other cases it would still be useful if IS could be turned ON/OFF before print (not mid print).

narno2202, 2023-12-21:
This is exactly that you can do with `FT_MOTION`.

Tupson444, 2023-12-21:
Great. As far as I can see the FT Motion includes Input Shaping, so this issue is solved with it then?

thisiskeithb, 2023-12-21:
> so this issue is solved with it then?

This "issue" is already solved since you disable Input Shaping by zeroing out the frequency / setting it to a low frequency that doesn't affect your print. That is how it was designed.

You disable other features in Marlin this way like Linear Advance. There's no on/off. You simply set it to 0 with an `M900 K0`.

Tupson444, 2023-12-21:
>by zeroing out the frequency

Yes I know, but I specifically suggested being able to turn it off without resetting frequencies (so you can later just turn it on, while the printer remembers previously set frequencies).

tombrazier, 2024-05-08:
We had a long conversation about how to turn IS on and off during the IS development. Setting frequency to zero won. A large part of the reason was that makes it consistent with other features in Marlin. There might have been other reasons too, can't recall. The conversation will still all be there on one of the early IS PRs, though. You'll need to engage those arguments if you want this change.

closed 2024-12-09 by thisiskeithb