MarlinFirmware/Marlin #26436: "No E stepper plug left for Z2!"

ysnuysl, 2023-11-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

I use the MKS Robin Nano V3.1 motherboard for my 1 CoreXY dual Z motor printer.

According to the information shared by the company, the software; I installed "https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware" from here and proceeded according to the guide.

Everything was working fine, but I couldn't manage to control the 2nd motor for Z with the 2nd driver. 2. I inserted the drive into the E1 slot and made the necessary software updates, referring to the video "https://www.youtube.com/watch?v=6UgT9YqY3UA". "No E stepper plug left for Z2!" Why am I getting the error? Your support is very important to me. I couldn't find a source :(

### Version of Marlin Firmware

marlin 2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-11-17:
MKS' fork is out of date and likely full of old bugs. I can compile with an MKS Robin Nano V3.1 that has five drivers enabled using current `bugfix-2.1.x` without issue.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-11-17 by thisiskeithb

ysnuysl, 2023-11-17:
When I tried bugfix-2.1.x, I could not find my motherboard in the list. I installed this first but it didn't work, why could that be?

thisiskeithb, 2023-11-17:
> When I tried bugfix-2.1.x, I could not find my motherboard in the list. I installed this first but it didn't work, why could that be?

`BOARD_MKS_ROBIN_NANO_V3_1`:

https://github.com/MarlinFirmware/Marlin/blob/19869d3a7db80b875e8d3706d3bbf1c62d3d1cb1/Marlin/src/core/boards.h#L443-L443

Please use the support links from my last comment since this is not a bug.

ysnuysl, 2023-11-17:
I don't understand how it could be possible, I'm immediately reinstalling 2.1 and trying it. I will give information.

thisiskeithb, 2023-11-17:
> I don't understand how it could be possible, I'm immediately reinstalling 2.1 and trying it. I will give information.

MKS Robin Nano V3.1 with five drivers compiles fine:

```prolog
Linking .pio/build/mks_robin_nano_v3_1/firmware.elf
Checking size .pio/build/mks_robin_nano_v3_1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.8% (used 18060 bytes from 131072 bytes)
Flash: [====      ]  43.4% (used 227760 bytes from 524288 bytes)
Building .pio/build/mks_robin_nano_v3_1/firmware.bin
rename_target([".pio/build/mks_robin_nano_v3_1/firmware.bin"], [".pio/build/mks_robin_nano_v3_1/firmware.elf"])
=============================== [SUCCESS] Took 60.44 seconds ===============================

Environment          Status    Duration
-------------------  --------  ------------
mks_robin_nano_v3_1  SUCCESS   00:01:00.443
================================ 1 succeeded in 00:01:00.443 ================================
 *  Terminal will be reused by tasks, press any key to close it.
 ```

Again, please use the support links from [my first comment](https://github.com/MarlinFirmware/Marlin/issues/26436#issuecomment-1816571904) for further troubleshooting.

thisiskeithb, 2023-11-17:
As the error states, you are using the incorrect build environment.

You are trying to use `mks_robin_nano_v3_usb_flash_drive_msc` when you should be using `mks_robin_nano_v3_1_usb_flash_drive_msc`.

ysnuysl, 2023-11-17:
I noticed after writing, I'm correcting it immediately :) thanks

ysnuysl, 2023-11-17:
Marlin 2.1 has changed a lot :)

As a result of the long compilation time; I got the error "c:\users\asus\desktop\yazilimlar\marlin2.1_mak-2\marlin\configuration_adv.h:1827:29: error: 'SD_DETECT_PIN' was not declared in this scope" :( This is the first time I encounter this.

thisiskeithb, 2023-11-17:
> This is the first time I encounter this.

Because your config is not correct. This compiles fine with flash drive support enabled:

```prolog
Linking .pio/build/mks_robin_nano_v3_1_usb_flash_drive_msc/firmware.elf
Checking size .pio/build/mks_robin_nano_v3_1_usb_flash_drive_msc/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  22.6% (used 29644 bytes from 131072 bytes)
Flash: [=====     ]  45.6% (used 239328 bytes from 524288 bytes)
Building .pio/build/mks_robin_nano_v3_1_usb_flash_drive_msc/firmware.bin
rename_target([".pio/build/mks_robin_nano_v3_1_usb_flash_drive_msc/firmware.bin"], [".pio/build/mks_robin_nano_v3_1_usb_flash_drive_msc/firmware.elf"])
===================================== [SUCCESS] Took 59.40 seconds =====================================

Environment                              Status    Duration
---------------------------------------  --------  ------------
mks_robin_nano_v3_1_usb_flash_drive_msc  SUCCESS   00:00:59.398
===================================== 1 succeeded in 00:00:59.398 =====================================
 *  Terminal will be reused by tasks, press any key to close it.
```

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server