MarlinFirmware/Marlin #26437: [BUG] Input Shaping doesn't work

informaticoantonio, 2023-11-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I printed more than 20 ring towers and I don see any difference between enabling input shape or not, even trying to increasing the F from 15 to 140 Hz. I have followed many guidelines to with every step, changing the PLA, changing the tension on the belts, reviewing each line in the firmware, changing from Prusa Slicer to Orca Slicer, changing the speed from 100, to 80 to 60 mm/s of print, disabled firmware retraction and Z hop, disable Linear Advance (from the beginning), disable fade height (but not ABL as I need it), reviewed motor current, max acceleration ... I follow the guide lines here:
https://marlinfw.org/docs/gcode/M593.html
https://support.th3dstudio.com/helpcenter/unified-marlin-input-shaping-tuning-guide/
https://www.youtube.com/watch?v=7VLuz3hReYw&t=21s&ab_channel=TeachingTech
https://www.youtube.com/watch?v=6zMNwiJX2CQ&ab_channel=ChrisRiley
And some more ...
I look into forums and all the mistakes that other people has I don't (or this is what I made sure I did not)
I'm using bugfix-2.1.x. and my configuration can be found here:
https://github.com/informaticoantonio/Marlin/blob/bugfix-2.1.x/Marlin/Configuration.h
https://github.com/informaticoantonio/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h

I'm desperate, I expend days and I just don't know what else to try ...
Any help, please, is this any short of bug in the firmware? Did I made some kind of dummy mistake? obviously something is wrong

Please, help!

### Bug Timeline

Is my first time trying Input Shaping

### Expected behavior

To find the correct Frequency for Input Shaping and see ghosting to be gone

### Actual behavior

It doesn't mother how many different ring tower I print that all looks the same and I can not find the right frequency to setup X and Y.
With or without input shaping is always the same ghosting 

### Steps to Reproduce

1. Enable:
#define INPUT_SHAPING_X
#define INPUT_SHAPING_Y
2. update de firmware to the printer
3. Init EEPRON (I got the right values directly configured in the Firmware except the ABL)
4. Check the Z off and print some test for bed levelling
5. Print a cube test to see if everything is fine
6. Load the ringing_tower into the Slicer
7. Follow all the steps for the configuration in the slicer (Prusa/Orca)
8. Download the Gcode and make sure that M593 increase for the layers
9. Sent to Octoprint and print
10. Review the tower looking for best value of F for X and Y
11. Nothing change 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Nov 16 2023 18:39:48)

### Printer model

Artillery SideWinder X1

### Electronics

BOARD_BTT_SKR_V1_4_TURBO

### LCD/Controller

BIGTREETECH TFT35-E3 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-11-17:
Come to the Marlin discord.

closed 2023-11-17 by ellensp

informaticoantonio, 2023-11-18:
As @Dust mentioned in Discord, the acceleration was too low, I changed to 5000 and now I can see the ringing.