MarlinFirmware/Marlin #26442: [BUG] Incorrect bed temp on Fystec 2.0 DGUS display

fortis852, 2023-11-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Chaotic bed temp (-900,-600, 50, 600) on DGUS_LCD_UI_FYSETC with firmware FYSTEC 2.0 on LCD. I changed line 395 in bed section DGUSDisplayDef.cpp like in heater section, and it works. 

`  #if HAS_HEATED_BED
    VPHELPER(VP_T_Bed_Is, &thermalManager.temp_bed.celsius, nullptr, ScreenHandler.DGUSLCD_SendWordValueToDisplay),`
to

`#if HAS_HEATED_BED
    VPHELPER(VP_T_Bed_Is, &thermalManager.temp_bed.celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>),`

https://github.com/MarlinFirmware/Marlin/blob/e45eddfd6c4008d82b3aefe4764793b0669dc60a/Marlin/src/lcd/extui/dgus/fysetc/DGUSDisplayDef.cpp#L395C1-L395C1

### Bug Timeline

Dont know

### Expected behavior

In octoprint and serial bed_temp is correct. I'd like to see it on my DGUS screen.

### Actual behavior

Chaotic bed temp (-900,-600, 50, 600) 

### Steps to Reproduce

1. See temp on DGUS FYSTEC 2.0 dispay

### Version of Marlin Firmware

02010201

### Printer model

custom

### Electronics

mks_gen_l_v2

### LCD/Controller

DGUS FYSTEC 2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

fortis852, 2023-11-18:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13401261/Marlin.zip)

thisiskeithb, 2023-11-20:
Applying your suggestion:

```c
VPHELPER(VP_T_Bed_Is, &thermalManager.temp_bed.celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>)
```
...to current bugfix, it becomes:

```c
VPHELPER(VP_T_Bed_Is, &thermalManager.temp_bed.celsius, nullptr, screen.SendFloatAsLongValueToDisplay<0>),
```

Can you test if this change works on current `bugfix-2.1.x`?

fortis852, 2023-11-24:
Yes, i will test

thisiskeithb, 2023-11-29:
Please report your findings in https://github.com/MarlinFirmware/Marlin/pull/26484!

closed 2024-05-25 by github-actions[bot]