MarlinFirmware/Marlin #26448: [BUG] ANET A8 1.5 motherboard requires Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW, yet #define BLTOUCH requires HIGH value??

PythonesqueSpam, 2023-11-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My ANET A8 1.5 motherboard requires Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW to work with a microswitch on Z-min.
I've tried to upgrade my printer with both a BL and then a CR Touch, but with each, #define BLTOUCH triggers an error for Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW stating only a HIGH value is acceptable.
I noted that when Z_MIN_PROBE_ENDSTOP_HIT_STATE is set to HIGH and #define BLTOUCH uncommented, the use of Gcode M119 shows that Z-min is always TRIGGERED.
This then seems to cause the BL/CR Touch probe to malfunction on initial probe deployment, triggering repeated probe deployments with an accompanying printer reboot cycle.
The reboot cycle requires TIOABOA to correct it.
These events occurred with the BL Touch I then returned believing it was faulty and the subsequent CR Touch I bought to replace it.

Speculating I suspect that, as in the case of a simple microswitch, LOW values for Z_MIN_PROBE_ENDSTOP_HIT_STATE and   Z_MIN_ENDSTOP_HIT_STATE may be required for the ANET A8 1.5 motherboard to correctly address the CR Touch device.

The CR Touch is currently wired with YELLOW to LCD pin 3 (Servo pin 27), WHITE to J3 pin 8 (GND), BLUE to Z-min pin 3 (Signal), BLACK to Z-min pin 2 (+5V) and RED to Z-min pin 1 (GND).  So to my knowledge the wiring config is correct. I've also used a meter to verify that the wiring is contiguous and that the WHITE and RED wires are on a common ground within the CR Touch device.

I'm using Marlin bugfix 2.1.x

### Bug Timeline

New

### Expected behavior

The probe would deploy (M401) and retract (M402) on respective GCode commands

### Actual behavior

The CR Touch probe malfunctioned on initial probe deployment, triggering repeated probe deployments with an accompanying printer reboot cycle.
The reboot cycle requires TIOABOA to correct it.

### Steps to Reproduce

Attach CR Touch probe using wiring and methods researched online; Wired with YELLOW to LCD pin 3 (Servo pin 27), WHITE to J3 pin 8 (GND), BLUE to Z-min pin 3 (Signal), BLACK to Z-min pin 2 (+5V) and RED to Z-min pin 1 (GND).
Update firmware with Z_MIN_PROBE_ENDSTOP_HIT_STATE,  Z_MIN_ENDSTOP_HIT_STATE, #define BLTOUCH, #define AUTO_BED_LEVELING_BILINEAR.
Send M401 Gcode to printer.

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

ANET A8

### Electronics

ANET A8 v1.5 motherboard

### LCD/Controller

ZONESTAR_LCD

### Other add-ons

BL or CR Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/13418115/Marlin.Configs.zip)

ellensp, 2023-11-21:
There is already one other board the K8400 that has a physical inverter on the probe pins
This  is added to their pins_K8400.h to over come this. 

#if ANY(BLTOUCH, TOUCH_MI_PROBE)
  #define INVERTED_PROBE_STATE
#endif

But only if you can confirm there is a electrical device on the board  inverting it somewhere, as the probes are all standard and all trigger the same way

ellensp, 2023-11-21:
This is your board ?

[image]

There is nothing obvious that would invert anything, there are two chop, one is the usb/serial converter, the other converts to 3.3v for the sd card

ellensp, 2023-11-21:
I suspect your hitting this issue
https://www.antclabs.com/wiring3-1

closed 2024-03-18 by github-actions[bot]