MarlinFirmware/Marlin #26451: [BUG] No warning on >100% RAM usage

non-bin, 2023-11-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My original title for this was "Compiling with FT_MOTION causes firmware to lockup during boot", until I noticed
`echo: Free Memory: -29396  PlannerBufferBytes: 1600` Negative free memory?
`RAM:   [==========]  193.5% (used 15849 bytes from 8192 bytes)` 193%?

<details> 
  <summary>Boot messages and full Build log</summary>

```
start
PowerUp
 Brown out Reset
 Watchdog Reset
 Software Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2023-11-21 | Author: (Alice, Creality Ender-5 Plus)
echo: Compiled: Nov 22 2023
echo: Free Memory: -29396  PlannerBufferBytes: 1600 <------------------------------- Negative free memory?
//action:notification Print Paused
echo:SD card released
echo:busy: processing
```

Then it locks up to the point were disconnecting and reconnecting serial console doesn't reset it, it has to be powered off and on

```
Executing task: C:\Users\jacka\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing mega2560 (board: megaatmega2560; platform: atmelavr@~4.0.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (4.0.1) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr @ 5.1.0
 - tool-avrdude @ 1.60300.200527 (6.3.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 1.0
|-- SPI @ 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\registers.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\babystep.cpp.o
[... 198 lines cut ...]
Stock Creality V2.2 board

### LCD/Controller

DGUS Reloaded

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/13426468/config.zip)

EvilGremlin, 2023-11-21:
This is not a bug, compiler/linker don't throw an error on RAM overflow. Won't be of much use because not everything is accouted for and you need to have some RAM free (500B at least)
Not sure if we can parse terminal output and throw error with script, probably no.

non-bin, 2023-11-21:
Fair enough, maybe I'll report it to pio then?

EvilGremlin, 2023-11-21:
It's not really pio either, it's gcc area. But pio could implement parse&error indeed.

thisiskeithb, 2023-12-03:
Fixed upstream in PlatformIO:

- https://github.com/platformio/platformio-core/issues/4791