MarlinFirmware/Marlin #26458: [FR] UBL Autofill Lower than Closest Value (Option)

alfondehesa, 2023-11-24:
### Is your feature request related to a problem? Please describe.

My CR10 bed is so warped that, with my 45 mm probe X offset, assuming the Z offset of the edge of the bed to be the same as the closest probed point (45 mm in) results in too high first layers.

### Describe the feature you want

Currently, Marlin UBL can auto-autofill points which either interpolates the height of the unreachable points if they are trending upwards (will raise Z offset if necessary). Otherwise, if the interpolated Z offset is **lower** than the closest point, it will default to using the closest value in an abundance of caution, so as to not drive the nozzle too far into the bed and damage it.

There should be an option in Marlin's configuration file (with proper warnings) to allow UBL autofill to infer z offsets when their interpolation results in lower values than the closest points.

stef-evotech, 2023-12-20:
Hi! 

Sry I don't quite get what you're after but I am assuming you need control over how the mesh points are treated which are not directly measured. If you are using UBL thats what G29 P3.x is for. I'm citing a comment from the source code: 

                // P3.1  use least squares fit to fill missing mesh values
                // P3.10 zero weighting for distance, all grid points equal, best fit tilted plane
                // P3.11 10X weighting for nearest grid points versus farthest grid points
                // P3.12 100X distance weighting
                // P3.13 1000X distance weighting, approaches simple average of nearest points 

additionally P3 or P3.0 should default to something referred as "smart_fill", which I haven't had the time to investigate further yet. Now the thing is what to choose, and from my experience you either have a bed thats basically a plane but a noisy sensor, For that I would start testing P3.1 or P3.10. A distinct cup or bow in your bed could mean that your are better of with P3.13 or switiching over to ABL. And for cases like mostly a plane but your edges are "feathered" up or down in a region your sensor cannot reach, well, I figure there is no automated solution or a solution where an extrapolation of the measured points makes sense.

alfondehesa, 2023-12-26:
@stef-evotech Hi!

I see... I found the same on the code. I guess I was a bit confused on its inner workings. On Marlin docs online, I see P3.XX is not mentioned, but instead they refer to this: