MarlinFirmware/Marlin #26468: [BUG] Mega1284p based devboard freeze up, not responding to encoder.

syriloak, 2023-11-26:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Description:
Marlin freeze up after half a minute. Not responding to Encoder, not reading temp.

Board:
I've designed my own board.
MCU: Atmega1284P tested with 8MHz int. osc 16,20,32MHz external crystal.
Pin assignment: 
-PA0(ADC0) : Hotend sensor(4k7 to 5V, 100k sensor)
-PA1(ADC1): Heated bed sensor (4k7 to 5V, 100k sensor)
-PA2: X Limit switch (4k7 to GND)
-PA3: Y Limit switch (4k7 to GND)
-PA4: Z Limit switch (4k7 to GND)
-PA5: Inductive or Capacitive sensor (4k7 to GND)
-PA6: SD-Detect
-PA7: SD-SS

-PC0: I2C SCL (4k7 to 5V) for SSD1306 OLED
-PC1: I2C SDA (4k7 to 5V) for SSD1306 OLED
-PC2: Z DIR (4k7 to GND)
-PC3: Z STP (4k7 to GND)
-PC4: Y DIR (4k7 to GND)
-PC5: Y STP (4k7 to GND)
-PC6: Z DIR (4k7 to GND)
-PC7: Z STP (4k7 to GND)

-PD0: RXTTL (for RS232-USB)
-PD1: TXTTL (for RS232-USB)
-PD2: ENA E (4k7 to GND)
-PD3: ENA Z (4k7 to GND)
-PD4: ENA Y (4k7 to GND)
-PD5: ENA X (4k7 to GND)
-PD6: E DIR (4k7 to GND)
-PD7: E STP (4k7 to GND)

-PB0: BTN-ENC (4k7 to 5V) Encoder button
-PB1: BTN-EN1 (4k7 to 5V)  Encoder A
-PB2: BTN-EN2 (4k7 to 5V) Encoder B
-PB3: Heated bed PWM (OC0A)
-PB4: Hotend PWM (OC1A)
-PB5: MOSI (for SD CARD)
-PB6: MISO (for SD CARD)
-PB7: SCK (for SD CARD)

To prevent false bug report PB0,PB1 and PB2 pins were tested with a small C program, the MCU is not faulty.

pins.h modified to allow custom motherboard selection:

#elif MB(XVORTEX)
#include "sanguino/pins_XVORTEX.h" 
// ATmega644P, ATmega1284P
env:sanguino1284p_optimized env:sanguino1284p env:sanguino644p

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/13467771/Archive.zip)

### Bug Timeline

After the first compile. No previous history.

### Expected behavior

Expected to not freeze up after ~30sec. 
After the boot screen's shord delay the status screen showing up, but unable to enter menu, fw not responding to encoder at all. 

### Actual behavior

Unable to enter menu, not responding to encoder, freeze up.

### Steps to Reproduce

Compile, upload, be sad.

### Version of Marlin Firmware

Unable to enter menu. According to the download page it is v2.1.2.1

### Printer model

Carthesian, unique.

### Electronics

Unique.

### LCD/Controller

SSD1306 I2C 128x64 OLED

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Currently no bed leveling is enabled, but planning to do it his issue resolved.

ellensp, 2023-11-27:
Your pins 

You have conflicts on pin 0 and pin 1

Even if you never turned on the heaters, the firmware still initializes the pins
BTN_EN1 is an input
HEATER_0_PIN s an output
Which ever one was set last will be what it is set to

```
/**
 * PIN:   0   Port: B0   BTN_EN1
 *                       HEATER_0_PIN
 * PIN:   1   Port: B1   BTN_EN2
                         HEATER_BED_PIN
 * PIN:   2   Port: B2   BTN_ENC
 * PIN:   3   Port: B3
 * PIN:   4   Port: B4   SDSS
 * PIN:   5   Port: B5   AVR_MOSI_PIN
 * PIN:   6   Port: B6   AVR_MISO_PIN
 * PIN:   7   Port: B7   AVR_SCK_PIN
 * PIN:   8   Port: D0   RXD
 * PIN:   9   Port: D1   TXD
 * PIN:  10   Port: D2   E0_ENABLE_PIN
 * PIN:  11   Port: D3   Z_ENABLE_PIN
 * PIN:  12   Port: D4   Y_ENABLE_PIN
 * PIN:  13   Port: D5   X_ENABLE_PIN
 * PIN:  14   Port: D6   E0_DIR_PIN
 * PIN:  15   Port: D7   E0_STEP_PIN
 * PIN:  16   Port: C0   SCL
 * PIN:  17   Port: C1   SDA
 * PIN:  18   Port: C2   Z_DIR_PIN
 * PIN:  19   Port: C3   Z_STEP_PIN
 * PIN:  20   Port: C4   Y_DIR_PIN
 * PIN:  21   Port: C5   Y_STEP_PIN
 * PIN:  22   Port: C6   X_DIR_PIN
 * PIN:  23   Port: C7   X_STEP_PIN
 * PIN:  24   Port: A7   TEMP_0_PIN
 * PIN:  25   Port: A6   TEMP_BED_PIN
 * PIN:  26   Port: A5   X_STOP_PIN
 * PIN:  27   Port: A4   Y_STOP_PIN
 * PIN:  28   Port: A3   Z_STOP_PIN
 * PIN:  29   Port: A2   Z_MIN_PROBE_PIN
 * PIN:  30   Port: A1
 * PIN:  31   Port: A0
 */

```

syriloak, 2023-11-27:
Hi ellensp!

Thank you very much for your effort and sorry for wasting your time.

I'have tested this stuff in every possible way, and it still not working properly.
Then I've realized that the HAL for AVR is based on this "wonderful" arduino thing.
From this point it's simply doesn't worth my time and energy, and I don't want to waste anyone's time.
This "duino" stuff is something that I'd like stay as far as humanly possible.

I wish you a wonderful and productive day!

closed 2023-11-27 by syriloak