MarlinFirmware/Marlin #26474: [FR] Support for Z-Axis Input Shaping in the M593 G-code

JaneX8, 2023-11-28:
### Is your feature request related to a problem? Please describe.

Currently Marlin does not offer accelerometer support in regards to Input Shaping. However, I think that the KUSBA (https://github.com/xbst/KUSBA) USB accelerometer (or similar USB device) **connected to the printer host** is the most accessible (plug-and-play) way to bring Input Shaping to most Marlin printer users. This is why I created an issue in the repo of OctoPrint (a host) and not the Marlin repo here: https://github.com/OctoPrint/OctoPrint/issues/4912.

However, Marlin needs to support it and as far as I know it only supports Input Shaping for the X and Y axes (https://marlinfw.org/docs/gcode/M593.html).

Ideally, I would like to hookup two KUSBA devices to my Raspberry Pi, one on the extruder (for X and Y axes) and one under the bed (for Z axis). However, even if OctoPrint implements support, Marlin does not support Z-axis Input Shaping.

### Describe the feature you want

Support for Z-Axis Input Shaping in the M593 G-code

oliof, 2023-11-28:
Z axis input shaping is not something that other firmwares do. KUSBA is also dependent on the Klipper multi MCU implementation.

One way to get measurements from an accelerometer is shown in https://www.youtube.com/shorts/OjjMlhe9nOs (unfortunately unpublished, TL;DW: accelerometer taped to moving part with processing and display of results on an m5stick)

JaneX8, 2023-11-28:
Interesting. Thanks for sharing. I am unsure what multi-MCU support is, but I don't understand why we need 'multi-MCU' in Marlin firmware. I suggest the calibration is done on **the host (being a microcontroller), so not Marlin**. Marlin has then to take the settings using M593 and M493. I'm only requesting Z-Axis Input Shaping support in Marlin Firmware. This way, Marlin should not have to deal directly with the accelerometer, usually having limited hardware to do so.

For example, OctoPrint, running on a printer host, could handle the accelerometer hardware and measurements and then send it to Marlin. I suggested that here: https://github.com/OctoPrint/OctoPrint/issues/4912.

oliof, 2023-11-28:
I guess I misunderstood your initial issue description then; the m5stick solution I showed does what you ask for (but you would still manually input the results).

I still don't understand why you say Z measurements are needed, input shaping is applied to fast moves in X and Y only, never on Z. The only reason you see 2 accelerometers on bed slingers is that you measure the Y on the bed and X on the nozzle (and this could be done by two separate runs with a single accelerometer as well).

kamskanagi, 2023-11-28:
Does Marlin support the GD32F303RET6?

Currently having an issue the Z-axis. We want to increase the Z-axis feedrate and rebuild the firmware. But cant find the board support for GD32F303RET6. 

Need urgent help on how to achieve this.

'Thanks

gudvinr, 2023-11-28:
> Z-Axis Input Shaping would allow for faster and smoother Z-axis movement

@JaneX8 Can you give an estimate of Z axis speed that you expect with Z-axis input shaping? What kind of issues do you personally have where you don't have enough speed and acceleration for Z?

With Z, it's either bed or toolhead that is moving and you can't move it fast because both of those are heavy. Realistically, even for non-planar prints distance in Z direction is so small (due to nozzle clearance) that you won't be getting speeds that are close to what you get in X and Y directions anytime soon.

JaneX8, 2023-12-02:
I'm not sure I will be able to answer that. 

_For me personally, I use a Creality Ender 3 V2 with dual axis upgrade. The Z axis speed for that printer is also discussed here: https://www.reddit.com/r/3Dprinting/comments/10ehzf1/can_i_damage_my_ender_3_v2_if_i_increase_the_max/. I think the max mechanical speed is up to 150mm/s or 180mm/s. But that's probably not what you meant. However, I'm planning to experiment with non-planar prints, maybe upgrading my printer if needed._

Again, I know nearly enough about Input Shaping, I just suggested an idea to get support in Marlin because that's basically the fundament on which many build.

I think KUSBA or any other USB accelerometer, connected to a host of a printer would make Input Shaping very accessible to many even budget printers, and low entry because it doesn't require any soldering or wiring. All I'm requesting here is to consider supporting Input Shaping on all axes (and take the values from the host using some G-code), Marlin is used on a huge variety of printers so my personal acceleration speeds won't matter much.

gudvinr, 2023-12-02:
> I think KUSBA or any other USB accelerometer, connected to a host of a printer would make Input Shaping very accessible to many even budget printers

For that you need printer to act as USB host, which is not possible for (probably) most Marlin-based printers

JaneX8, 2023-12-02:
To clarify by 'Printer Host' I meant for example a computer or Rasbperry Pi (OctoPi/OctoPrint) that controls the printer. In my case I print everything through OctoPi, which controls the printer over USB from the Raspberry Pi to the Creality Ender 3 v2 printer. Clearly the printer itself wouldn't have capabilities for adding USB devices, or perhaps even any accelerometer. So, in this case I'm thinking to connect a device like KUSBA to my Raspberry Pi, which can then use it to do Input Shaping calibration on demand, as it can control the printer and it can read the accelerometer, automated.

This should theoretically already be possible which is why I requested that here: https://github.com/OctoPrint/OctoPrint/issues/4912#issuecomment-1834039121. The point of this issue in the Marlin repo is simply for Z-Axis Input Shaping.

dbuezas, 2024-04-16:
I have a situation that would benefit from input shaping in Z:
My Ultimaker 2 (replaced 32 bit board) moves the bed up and down and the bed is supported only from the back. It is very boncy.

**Specifically when I use z hop (0.1mm), there is clear ringing after unretractions.** 
I have to lower z acceleration to 50mm/s² and Jerk to 1 to avoid that. This works but it makes xy travel moves stutter in the first layers because the z axis is working to follow bed levelling.

I initially thought it was some sort of intermittent VFA, but this 1 layer concentric pattern I made shows clearly that the issue is actually on unretractions.

Needless to say, the artefact piles up on layers and results in holes in the walls of the print.

[image]

[Concentric pattern.3mf.zip](https://github.com/MarlinFirmware/Marlin/files/15001498/Concentric.pattern.3mf.zip)

Sliced with Orca slicer, these settings are important:
[image]

dbuezas, 2024-04-16:
I should mention that even at the extremely low accelerations and jerks, this artefact is still visible with the light right, and it results in a VFA like defect

dbuezas, 2024-04-17:
I should also mention that the same effect happens on each layer change regardless of zhop, it is just more time consuming to print tests

tombrazier, 2024-05-06:
@dbuezas That image and your analysis is really interesting to see. @JonBr306 is working on Z axis input shaping. Do you fancy helping to test it?

dbuezas, 2024-05-06:
Sure thing. I think this pattern may be useful to tune the parameters

thisiskeithb, 2024-05-06:
> @JonBr306 is working on Z axis input shaping. Do you fancy helping to test it?

I’ll have a CoreXZ machine built here soon that would probably be a good test for this as well.

dbuezas, 2024-05-07:
> I have been wondering about that. I was thinking of a variation of https://marlinfw.org/tools/input_shaping/freq-calibr.html.

Yes! that calibration is fantastic.

A bunch of straight lines (each starting with a z hop) should would do the trick for z IS

tombrazier, 2024-05-07:
> Yes! that calibration is fantastic.

Thank you. It was my idea.

thisiskeithb, 2024-05-09:
Please test & report findings in https://github.com/MarlinFirmware/Marlin/pull/27073

closed 2024-05-20 by thinkyhead