MarlinFirmware/Marlin #26476: Ender 3 v2 compile failed with ProUI

dmitrygribenchuk, 2023-11-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello. When I try to build firmware for Ender 3 v2 (STM32F103RE_creality) with ProUI enabled (#define DWIN_LCD_PROUI) I have an error:

Compiling .pio/build/STM32F103RE_creality/src/src/sd/SdBaseFile.cpp.o
Marlin/src/lcd/e3v2/proui/dwin.cpp: In function 'void dwinPrintAborted()':
Marlin/src/lcd/e3v2/proui/dwin.cpp:1725:3: error: 'hostui' was not declared in this scope
 1725 |   hostui.notify("Print Aborted");
      |   ^~~~~~
Compiling .pio/build/STM32F103RE_creality/src/src/sd/SdFatUtil.cpp.o
*** [.pio/build/STM32F103RE_creality/src/src/lcd/e3v2/proui/dwin.cpp.o] Error 1

### Bug Timeline

Last success build was about 2 weeks ago

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 v2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.h.gz](https://github.com/MarlinFirmware/Marlin/files/13486261/Configuration.h.gz)
[Configuration_adv.h.gz](https://github.com/MarlinFirmware/Marlin/files/13486262/Configuration_adv.h.gz)

ellensp, 2023-11-28:
https://github.com/MarlinFirmware/Marlin/pull/26308 didn't check if HOST_PROMPT_SUPPORT is enabled or not, it just assumed that it was

It also added the string "Print Aborted" when have MSG_PRINT_ABORTED already exists and supports multiple languages 

The fix is really simple

```DIFF
diff --git a/Marlin/src/lcd/e3v2/proui/dwin.cpp b/Marlin/src/lcd/e3v2/proui/dwin.cpp
index 366ecffbd9..8d4aa9106b 100644
--- a/Marlin/src/lcd/e3v2/proui/dwin.cpp
+++ b/Marlin/src/lcd/e3v2/proui/dwin.cpp
@@ -1722,7 +1722,7 @@ void dwinPrintAborted() {
       );
     }
   #endif
-  hostui.notify("Print Aborted");
+  TERN_(HOST_PROMPT_SUPPORT, hostui.notify(GET_TEXT_F(MSG_PRINT_ABORTED)));
   dwinPrintFinished();
 }

```

ellensp, 2023-11-28:
I have added a PR to fix this

closed 2023-11-28 by ellensp