MarlinFirmware/Marlin #26479: [BUG][v2.1.2.1] Impossible to load filament (LCD, M701)

ESP4Ever, 2023-11-28:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

When "Filament Load" button tapped, the extruder pulls around 20 mm of filament at very low speed and then "Paused" menu pop-ups. Tap on any button closes the window and nothing happening then, all other actions are blocked by message that filament load in progress. I seen such issue were already closed in year 2020 but I have this issue on the latest FW.

[image]
[image]

E-Step calibration and other extruder move options are working fine.

Config:
```
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     50  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      500  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH    20  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE  50  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH   450  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion.
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.
  #define ADVANCED_PAUSE_RESUME_PRIME          0  // (mm) Extra distance to prime nozzle after returning from park.
  //#define ADVANCED_PAUSE_FANS_PAUSE             // Turn off print-cooling fans while the machine is paused.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_PURGE_RETRACT       15  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_PURGE_DELAY       5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.
  #define FILAMENT_UNLOAD_PURGE_FEEDRATE      25  // (mm/s) feedrate to purge before unload

  #define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.
  //#define FILAMENT_CHANGE_RESUME_ON_INSERT      // Automatically continue / load filament when runout sensor is triggered again.
  //#define PAUSE_REHEAT_FAST_RESUME              // Reduce number of waits by not prompting again post-timeout before continuing.

  //#define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // If needed, home before parking for filament change

  #define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  #define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
```

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom

### LCD/Controller

BBT Octopus Max EZ controller + TFT35 v3.0.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ESP4Ever, 2023-11-28:
Thank you for your prompt reply!
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

I will... try

> You did not attach your configs like the Bug Report template asked for, but did you follow the [Marlin Dependencies section of BTT's TFT firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies)?

Yes, I enabled the options I'm using. Here is my config:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13491402/Marlin.zip)

> Does this happen when the TFT is in "Marlin Mode"?

I'm not using this mode. Currently I'm finalizing my Da Vinci 3-in-1 mod and only learning Marlin

thisiskeithb, 2023-11-28:
> I will... try

> I'm not using this mode. Currently I'm finalizing my Da Vinci 3-in-1 mod and only learning Marlin

Please try these & report back.

Also, be sure to update the TFT to the latest firmware.

ESP4Ever, 2023-11-28:
> > I will... try
> 
> > I'm not using this mode. Currently I'm finalizing my Da Vinci 3-in-1 mod and only learning Marlin
> 
> Please try these & report back.
Unfortunately it fails to build

```
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:304:11: error: missing binary operator before token "("
  304 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
```

> Also, be sure to update the TFT to the latest firmware.
It is on the latest available FW

thisiskeithb, 2023-11-28:
You have to port your config to the latest version, not just copy & paste your old one. The `EITHER()` macro no longer exists.

thisiskeithb, 2023-11-28:
>  Is the release note with breaking changes available?

There are no release notes for `bugfix` branches and `2.1.3` is not out yet.

ESP4Ever, 2023-11-28:
> > Is the release note with breaking changes available?
> 
> There are no release notes for `bugfix` branches and `2.1.3` is not out yet.

Need help here again... Now it doesn't compile because of the LCD controller.
I see that this section was heavily reworked and it is not clear which controller to select for TFT35 v3.0.1 (UART)

thisiskeithb, 2023-11-28:
> Need help here again... Now it doesn't compile because of the LCD controller.
> I see that this section was heavily reworked and it is not clear which controller to select for TFT35 v3.0.1 (UART)

Connect both 10-pin EXP cables & enable `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`.

Your config is also missing these required defines listed in the Marlin Dependencies link I posted above:

- `EMERGENCY_PARSER`
- `PARK_HEAD_ON_PAUSE`

thisiskeithb, 2023-11-28:
>  Besides, I got this error when I tried to use reprap controller

Ahh, right. `EMERGENCY_PARSER` is still required.

> Defined but i think it is optional

[image]

Not in the configs you posted.

ESP4Ever, 2023-11-28:
> Not in the configs you posted.
It is from v2.1.2.1 and it is perfectly works except filament loading)
Will try to enable EMERGENCY_PARSER there and check again.

Regarding bugfix-2.1:
If I try to use BTT_MINI... I'm getting tons of NeoPixel errors, which used as a case light and have different type

```
Compiling .pio\build\STM32H723ZE_btt\src\src\HAL\STM32\eeprom_sram.cpp.o
Compiling .pio\build\STM32H723ZE_btt\src\src\HAL\STM32\eeprom_wired.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:52,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2271:6: error: #error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."
 2271 |     #error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."
      |      ^~~~~
*** [.pio\build\STM32H723ZE_btt\src\src\HAL\STM32\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:52,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2271:6: error: #error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."
```

thisiskeithb, 2023-11-28:
>  If I try to use BTT_MINI... I'm getting tons of NeoPixel errors, which used as a case light and have different type

Yes, those errors are listed in the output: ` #error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."`

If you're not hooking up a `BTT_MINI_12864`, then there's no need for you to define it.

ESP4Ever, 2023-11-28:
> > If I try to use BTT_MINI... I'm getting tons of NeoPixel errors, which used as a case light and have different type
> 
> Yes, those errors are listed in the output: ` #error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."`
> 
> If you're not hooking up a `BTT_MINI_12864`, then there's no need for you to define `BTT_MINI_12864`.

If so then there is no option to compile this bugfix for my setup(((

thisiskeithb, 2023-11-28:
> If so then there is no option to compile this bugfix for my setup

Marlin will compile fine without a display, so just don’t define one.

Since you’re not able to connect your TFT via the two EXP cables, it’ll only communicate via serial (without “Marlin Mode) since it’s a standalone computer that runs its own firmware and talks to Marlin via G-codes when in “Touch Mode”.

ESP4Ever, 2023-11-28:
Enabling EMERGENCY_PARSER solved the issue - four eyes is always better than two))) Thank you for your help and sorry to disturbed.
Now I have some weird feed speeds, not what I defined so I have to investigate further

closed 2023-11-28 by thisiskeithb