MarlinFirmware/Marlin #26482: [FR] Input Shaper disabled at slower speed

tklus, 2023-11-29:
### Describe the feature you want

I have been playing with Input shaping trying to get it dialed. I am getting close to a good speed/accel freq settings and was curious how difficult it would be to add a cut off point for input shaping as it relates to speed. the idea is, I can print most everything fast with input shaping, but adjust the speed for the outer layer to a much lower speed for better quality. can we set a variable for speed that if the speed is below a specified value, it will disable input shaping for that outer layer. It seems as though input shaping does not adjust with speed so I am getting induced ghosting from input shaper over compensating when running and much slower speeds.

XDA-Bam, 2023-12-18:
Input shaping should not create artifacts when printing slower, as long as it is tuned correctly. At least I never noticed any on my machine and I've been printing a lot at 10 - 15 mm/s recently due to testing a 1 mm nozzle.

Apart from that, I don't think the option to turn off IS for lower speeds could be implemented in any reasonable form, since sections of a print which will always be slow - such as sharp corners - are the sections where you _want to_ use input shaping, if is is enabled. Therefore, a speed threshold would effectively cripple the feature.

If you have trouble with specific designs, you can always turn off IS using `M593 F0`. Or you can use a post processing macro to disable/enable it before and after a layer or maybe even before/after just the outer perimeters, but that may be a bit cumbersome to implement.

tombrazier, 2024-04-06:
Input Shaping's effect decreases with speed. It should not be introducing artifacts at lower speeds. Do you have a replicable example of this happening with IS?

thisiskeithb, 2024-06-02:
For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-06-02 by thisiskeithb