MarlinFirmware/Marlin #26485: [BUG] Touch screen cannot be calibrated

clifo, 2023-11-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The touchscreen calibration remains in an infinite loop. It continuously repeats the calibration cycle without the possibility of termination.
If I disable calibration and enable 
  #define TOUCH_CALIBRATION_X 17880
  #define TOUCH_CALIBRATION_Y -12234
  #define TOUCH_OFFSET_X -45
  #define TOUCH_OFFSET_Y 349
touch screen does not respond to touch.

### Bug Timeline

new bug

### Version of Marlin Firmware

Marlin 2.1 Bugfix Branch DISTRIBUTION_DATE "2023-11-30"

### Printer model

Two Trees BlueR V2

### Electronics

MKS robin nano V1.2

### LCD/Controller

MKS_ROBIN_TFT35

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/13516657/Marlin_config.zip)

thisiskeithb, 2023-11-30:
Duplicate of #26454

thisiskeithb, 2023-11-30:
This will be fixed in #26455

closed 2023-11-30 by thisiskeithb

thisiskeithb, 2023-11-30:
#26455 is now merged, so download the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) for the fix.