MarlinFirmware/Marlin #26491: [BUG] SD Card Not detected on STM32F103VE

therealalexschwartz, 2023-12-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello all,

I have a Longer LK4 with a stock STM32F103VT06 chip. I used the prebuilt Marlin Binary from [here](https://github.com/3d-printing-canada/Longer-LK1-LK2-BL-Touch-Firmware) but was unable to build from source and make changes after I noticed some bugs.

I'm attempting to port those changes to the 2.1.x version so I am able to run the latest and hopefully contribute back my findings. I have been able to get everything working except the onboard SD Card slot.

The printer fails to detect an SD card and when attached over USB console I see it echo out "No SD Card Detected". I can confirm my SD card works as when I fall back to the pre-compiled version I am able to print from the same SD Card.

I enabled SD Card support in the configurations as well as in the advanced config file specifying onboard. Ive attempted various things found in other issues but with no success. I am unable to find what the original author did to enable SD card support if its a HAL file or a custom pins.h they wrote but so far all my attempts have failed. Hoping maybe someone knows where I should be looking to emulate what they did.

Any and all help would be greatly appreciated as after this point I am hoping the whole printer will be functional.

I am using the maple build, as when I use the non maple build I get a boot loop.

Build Command: platformio run  --environment STM32F103VE_longer_maple

Thank you! 

### Bug Timeline

New

### Expected behavior

I expected the SD card slot to be detected and usable.

### Actual behavior

I am unable to mount or use an SD card as it is not detected and over console reports No SDcard on boot.

### Steps to Reproduce

1. Install the latest marlin bugfix on a Longer LK4

### Version of Marlin Firmware

bugfix-2.1.x 2023-11-29

### Printer model

Longer LK4

### Electronics

Stock: STM32F103VT06

### LCD/Controller

TS_V19

### Other add-ons

BLtouch Sensor

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13533108/Configuration.zip)
[platformio.ini.zip](https://github.com/MarlinFirmware/Marlin/files/13533575/platformio.ini.zip)

therealalexschwartz, 2023-12-06:
Not sure if this info is helpful but here is the terminal output I see when I attach it to octopi

`Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x66e25d70, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok
Send: N2 M21*18
Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Dec  5 2023 21:39:16) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Alfawise U20 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:1
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:0
Recv: Cap:EP_BABYSTEP:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Send: M155 S2
Recv: echo:No SD card
Recv: ok
Send: M27 S1
Recv: ok
Recv: ok
Recv: Not SD printing
Recv:  T:26.78 /0.00 B:25.44 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:27.08 /0.00 B:25.74 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing`

Wuerfelix, 2024-05-05:
Hey, I have the same problem! But with the new 2.1.2.0 build on a Alfawise U20+. Is there a Bugfix or ideas to solve the Problem?

Wuerfelix, 2024-05-26:
I found the problem. The #define ONBOARD_SDIO entry was in src\pins\. stm32f1\pins_LONGER3D_LK.hlonger.h missing

closed 2024-05-26 by thinkyhead