MarlinFirmware/Marlin #26504: [BUG] B1 se plus  bugfix-2.1.x have white screen.

doshiyo103, 2023-12-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I downloaded bugfix2.1.x from the website, applied it to the latest configuration file for B1 se plus, 
installed it on an SD card, but when I boot up, I get a white screen. 

I tried connecting with Pronterface to the computer, and that works fine. 
M503 can read, and M115 displays the firmware version, indicating a successful firmware installation. 
However, the white screen issue leaves me unsure of what to do :(

### Steps to Reproduce

1.Download bugfix2.1.x from Marlin Github and open it.
2.Download B1 se plus .h and adv.h from the Marlin Github configuration files.
3.Overwrite the bugfix2.1.x with B1 se plus .h and adv.h, then build.
4.Place firmware.bin on the SD card, boot up, and encounter a white screen.

### Version of Marlin Firmware

bugfix2.1.x 

### Printer model

B1 se plus

### LCD/Controller

BTT TFT35 SPI V1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

the file same from here Marlin Configuration
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13593348/Configuration.zip)

my LCD
BTT TFT35 SPI V1.1
[image]

thisiskeithb, 2023-12-07:
SPI TFTs are broken on STM32F1 and STM32F4 right now. This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/26052

doshiyo103, 2023-12-07:
Hello, thank you for your prompt response. 
I've attempted to overlay the file from #26052 onto bugfix-2.1.x. 
This time, the screen shows the "Marlin mode," and the system can boot up normally. 

However, it has given rise to another issue. Each step of the operation now feels delayed, 
as if there's a noticeable lag. Could you please check and confirm this issue?