MarlinFirmware/Marlin #26508: [BUG] Babystepping / Configuration->Z Probe offset not working

bipsendk, 2023-12-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When going into configuration menu and selecting the Z Probe offset - then I have a value of -0.033 listed.
Turning the knob on the display does not change the value (one time it flipped to -0.034 - but otherwise no action at all).
I can push the rotary encoder button, and then the menu will revert to the configuration menu.
Also, when double-clicking while printing to enter the babystepping menu during printing, the menu/page pops up - but the value cannot be adjusted (the display shows no change).

### Bug Timeline

Has popped sometime after mid November

### Expected behavior

Being able to adjust the Z-offset (babystepping) in the menu.

### Actual behavior

Unable to change Z-offset in that particular menu.
Also unable to adjust z-offset babystepping during print.

### Steps to Reproduce

Start printer
Go to configuration menu
Go to Probe Z Offset menu
try to turn knob to change value

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

CoreXY (home built)

### Electronics

Bigtreetech SKR 1.3

### LCD/Controller

Fysetc Mini 12864 ver 2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13601400/Marlin.zip)

bipsendk, 2023-12-07:
Just checked - my firmware from November 28th has the menu working

ellensp, 2023-12-07:
@bipsendk that has already been fixed https://github.com/MarlinFirmware/Marlin/pull/26497

Get latest bugfix

closed 2023-12-07 by ellensp