MarlinFirmware/Marlin #26515: [FR] Inversed Markforged / Hybrid Corexy

jesterhead82, 2023-12-08:
### Is your feature request related to a problem? Please describe.

Currently Marlin supports only Markforged / Belt paths as defined in #23163
Klipper e.g. has the same issue. the implemention of a inversed belt (e.g. Ratrig Idex) is not in mainline yet.

As a workaround I changed the maths (replace + with - and vice versa) to support my printer (its not finished yet, but at least axis are moving now correctly).
I Changed only the maths for the XY type, but I guess the same could be usefull for YX layouts.

In theory, if somebody tells me how would such a feature would be properly named / configured I could try to assemble a PR...

Planner.cpp: 
~line 1992

 #elif ENABLED(MARKFORGED_XY)
    dm.a = (dist.a - dist.b > 0);               // Motor A direction
    dm.b = (dist.b > 0);            

~line 2065
 ABS(dist.a - dist.b), ABS(dist.b), ABS(dist.c)

~line 2113
  dist_mm.a = (dist.a + dist.b) * mm_per_step[A_AXIS];

Stepper.cpp
~Line 3290

  count_position.set(spos.a + spos.b, spos.b, spos.c);

~ 3383
 #elif ENABLED(MARKFORGED_XY)
      axis == CORE_AXIS_1
        ? count_position[CORE_AXIS_1] + count_position[CORE_AXIS_2]
        : count_position[CORE_AXIS_2]

### Are you looking for hardware support?

Custom Markforged kinematics

### Describe the feature you want

Possibility to configure inverse markeforged (different beltpath)

EvilGremlin, 2023-12-08:
MARKFORGED_XY & MARKFORGED_YX looks ok

thisiskeithb, 2023-12-12:
Closing since you've opened a PR.

PR #26516 (2023-12-13): Markforged inversed beltpath support (#26515)
My costum printer with "Markforged" Kinematics has a different Beltpath (X Carriage is fixed on the other side of the X Belt), compared to the already support markforged kinematics type introduced by #23163 

Something like that PR should fix #26515 

Hopefully I´ve got everything right, i was only able to test Markforged_XY, but I guess the same could be usefull for YX variant.

BTW. im not familiar with Marlins coding style or Github handling at all, so please be kind ;)
closed 2023-12-12 by thisiskeithb