MarlinFirmware/Marlin #26520: [FR] Avoid going beyond Z-MIN endstop when Z-MAX homing is done through a probe

dbuezas, 2023-12-11:
### Is your feature request related to a problem? Please describe.

In my Ultimaker 2 the bed moves up and down. When the steppers are disabled, the bed falls by gravity to the bottom.
I am using a BL-Touch acting as a Z-MAX probe, so when I home the axes, the printer adds z clearance BEYOND the existing Z-MIN endstop, and it skips steps and makes ugly noises.

As a workaround I left a sponge below the bed so it doesn't fall as deep. 

### Describe the feature you want

That the downward z movement is clipped once the Z-MIN stop is reached (particularly if it is already triggered), even when there is a z-max probe in use

### Additional context

The probe and endstops use independent pins, and they are both configured. I confirmed they are both in working order via an endstop test in the UI.

EvilGremlin, 2023-12-11:
You saying it backwards, your probe is Zmin and endstop is Zmax. Looks like you just need to reduce Z_MAX_POS a bit.

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN" "http://www.w3.org/TR/REC-html40/strict.dtd">
<html><head><style type="text/css">
p, li { white-space: pre-wrap; }
</style></head><body style=" font-family:'DejaVu Sans'; font-size:12pt; font-weight:400; font-style:normal;">

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: Download Marlin bugfix-2.1.x to test with the latest code. Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware. Connect to your printer from host software such as Cura, Printrun or Repetier Host. Send M502 and M500 to ensure your Configurations are applied. Issue the command M111 S247 to enable maximum logging. Perform a G28 to do your standard homing procedure. Do a G29 to probe the bed. This will also enable bed leveling. Do some of the moves that revealed problems before. Take notes. Copy the log output into a .TXT file and attach it to your next reply. Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.
--

</body></html>

EvilGremlin, 2023-12-11:
Also already exist - ENDSTOPS_ALWAYS_ON_DEFAULT, SD_ABORT_ON_ENDSTOP_HIT

closed 2023-12-11 by thisiskeithb

dbuezas, 2023-12-11:
Oh, that won't abort the print? Cool Thanks

dbuezas, 2023-12-12:
By the way, my printer moves the bed in the z axis, not the nozzle, so Z=0 is when the bed is up. That's why I said the min stop is the probe.

For reference, the discord thread where I first asked about this is here: https://discord.com/channels/461605380783472640/491165528295997450/1182757755812061235

ellensp, 2023-12-12:
But you didn't, you said and I quote "am using a BL-Touch acting as a Z-MAX probe"

dbuezas, 2023-12-13:
Oh darn, you are right!