MarlinFirmware/Marlin #26523: [FR] MMU3 Interface

eoyilmaz, 2023-12-12:
### Is your feature request related to a problem? Please describe.

I started working on the MMU3 interface, and I wanted to report this feature request to consolidate all future efforts in one place. So, anybody interested please feel free to join the discussion here.

I already have an MMU3 setup, albeit working on an Ender 3 Pro + SKR 3 EZ board with the 1.0.6 firmware of the MMU2s. Just updated the MMU firmware to version 3.0.2 today and I had to adjust some StalGuard thresholds and motor currents so the Idler homes properly. The communication between the boards is not working right now. My setup includes an Arduino Mega 2560 board inbetween the Marlin and MMU boards so that I can intercept the messages between these two boards more easily. The MMU code base is very clean and very easy to read, and it is surprisingly easy to build (I didn't expect that tbh).

So, yeah, let's do it!

### Are you looking for hardware support?

Yes, I'm interested in MMU3 support.

### Describe the feature you want

Fully working MMU3 with the advertised:

- Full two-way communication with the printer,
- LCD screen integration (status messages, real-time information, error codes)

ellensp, 2023-12-13:
MMU3 Information https://help.prusa3d.com/article/registers-mmu-mmu3_511780

eoyilmaz, 2023-12-13:
@ellensp thanks for the info...

Status update, I was able to communicate with MMU3 last night... The CRC8 stuff is working properly...

For prototyping purposes, I'm currently filtering out the CRC8 stuff from the MMU3 and add back to the commands coming from Marlin...

Next, what I want to do is to obtain the Prusa firmware (the currently shared source code I think is still communicating with MMU2s version 2.1.9, there is a MMU300 branch to be investigated) and see how different it is from the current implementation in Marlin.

plampix, 2023-12-13:
@eoyilmaz Anything above 3.13 will only work with MMU3 FW 3.0 and above. 3.12 and below will only work with 1.06 and below.

eoyilmaz, 2023-12-13:
We should also consider doing the same to be honest, at least we should remove the old MMU stuff completely.

thisiskeithb, 2023-12-13:
> we should remove the old MMU stuff completely.

To prevent breaking builds for users who upgrade Marlin, but run an older MMU/MMU firmware, we’d just need to add notes to the MMU config section with compatible MMU & MMU firmware versions.

eoyilmaz, 2023-12-13:
@thisiskeithb sure, but I was reffering the MMU1 stuff, the oldest version that nobody is probably not using anymore. Of course, we should keep everything related to MMU2/MMU2s.

> @eoyilmaz Anything above 3.13 will only work with MMU3 FW 3.0 and above. 3.12 and below will only work with 1.06 and below.

sure thing, I'll base my code to 3.13...

From Marlin's perspective, MMU with a firmware version of 1.0.6 is considered the `MMU2s` (you need to set the `MMU_MODEL` to `PRUSA_MMU2S` to work with it) and anything new (3.x.x) should be considered MMU3 (most probably you need to set the `MMU_MODEL` to `PRUSA_MMU3`), I don't remember if Prusa has ever released the MMU firmware 2.x stuff, from the commit messages I see that they renamed the 2.1.9 to 3.0.0.

thisiskeithb, 2023-12-13:
> sure, but I was reffering the MMU1 stuff, the oldest version that nobody is probably not using anymore. 

We don't remove support because something is old. There's no reason to remove it.

eoyilmaz, 2023-12-13:
> We don't remove support because something is old. There's no reason to remove it.

ah, sure then, it was already too early to talk about this tbh... let's add the MMU3 support, then we can talk about the rest 👍

eoyilmaz, 2023-12-13:
Instead of deep diving into the work directly, I would like to approach this in a more structured and professional manner, create an EPIC, investigate it with a SPIKE, divide the work in small manageable chunks etc. as we normally do in our studio. So I'm moving back to my own repository, will create all the work tickets there, I'll update this ticket most probably with the list of feature tickets that needs to be implemented and create PRs as I gone through them.

eoyilmaz, 2023-12-15:
Here is some progress on destructing the new protocol: https://github.com/eoyilmaz/Marlin/issues/2#issuecomment-1857207374

eoyilmaz, 2023-12-28:
Hey folks, let me do some progress reports:

- I moved all the MMU3 related source code from PrusaFirmware to Marlin, most of the code didn't require too much modification. But, as Prusa firmware is mostly based on Marlin1 I had to do a lot of detective stuff and find similar functionalities in Marlin2 and there were couple of places where the implementation were just existing in Prusa and not directly in Marlin.
- I had to comment out a lot of EEPROM related functionality coming from Prusa as I'm not experienced with reading & writing from PersistentStore and creating new data structures for that purpose and I need supervision. In PrusaFirmware, they have defined the addresses and the data sizes in an header file contrary to Marlin's way of laying out the data structure.
- Implemented (or moved/modified) the new M-Codes (M704, M705, M706, M707, M708, M709) as they were just wrappers to the new MMU2 API. Reading/writing to EEPROM stuff is currently disabled. So for now, for example the cutter is always enabled.
- For now, I had to disable some of the menus coming from Prusa as the menu system they are using is completely different, but I think I can handle these, but it will require a complete rewrite.
- I temporarily kept the MMU2/MMU2s menus and connected them to MMU3 api if the `MMU_MODEL` is `PRUSA_MMU3`.
- Ah yeah, and we now have a new `PRUSA_MMU3` value for `MMU_MODEL` that enables the MMU3 stuff to be compiled.
- I kept the MMU2/2s code intact, setting the `MMU_MODEL` to `PRUSA_MMU2/PRUSA_MMU2S` will compile Marlin with MMU2/S features.

Otherwise, I was just able to compile Marlin with MMU3 support and will test it after my current print finishes and I'll report back... I'm not expecting it work out of the box tbh.

Yeah, that's all for now, cheers...

eoyilmaz, 2023-12-28:
Surprise surprise 😄 it is mostly working 🎊 

I had to fix the required MMU firmware version in Marlin from 3.0.1 to 3.0.2. As I see PrusaFirmware requires an exact match of the MMU firmware otherwise disables the MMU communication after reporting something like "MMU requires FW Update disabling" or so.

Things are working:
- Marlin <-> MMU initialisation and communication
- The new M70x gcode commands
- MMU status and progress messages are properly displayed on the LCD, so when it loads a filament you can properly see a "Feeding to FINDA" etc. kind of messages in the status screen as it progress through all the stages of loading/unloading a filament.

Things are broken:
- Marlin menus are not working properly after displaying MMU related messages.
- I cannot get any MMU registry values with M707 A00x00 commands.
- Requesting a filament change through T0 is not fully working, but I think it is related to my setup, and the command is properly sent to MMU and a proper response is retrieved back, only the filament is not loaded properly.

anyways, I'm very excited as I was not expecting anything to function not even a little 😄

eoyilmaz, 2023-12-31:
Another progress report:

I just completed my first successful print with the MMU3 that had 36 filament changes (very small test print)... The main functions are now fully working, and I'm quite happy with my progress... I still need to work on the LCD reporting & interaction and EEPROM & PersistentStore stuff though... 

You can track my todo list here btw: https://github.com/eoyilmaz/Marlin/issues/3

eoyilmaz, 2024-01-03:
Status Update:

Just finished a 540+ filament change print job without any issues related to the Marlin + MMU3 (the only 2 issues I had were related to one of the filaments and my own setup):

[image]
[image]
[image]

eoyilmaz, 2024-01-03:
I also worked on the LCD and EEPROM related stuff and it is theoretically done. I'll test and report back. I'm very close to creating a PR 👍

eoyilmaz, 2024-01-04:
Latest changes:

- Updated MMU Menu:
  - Added Enable/Disable Cutter (this has 3 states, `0: Disabled`, `1: Enabled` `2: Always`, I kept off the state `2: Always` from the menus as even Prusa deems this feature as experimental).
  - Added "Cut Filament" menu if `MMU_HAS_CUTTER` is set in the `Configuration_Adv.h` and the Cutter is enabled.
  - Added Enable/Disable Spool Join feature (This still requires `M600 AUTO` to be implemented).
  - Added "Fail Stats" menu.
    - Added stats menu for the last print job
    - Added stats menu for the total failures of printers life time.
    - Added stats for total material changes done
- Updated ``settings`` to store the following data in EEPROM:
  - MMU print & fail stats
  - Cuter Enable/Disable state
  - Spool Join Enable/Disable state
  - MMU Stealth Mode (currently appears to be read-only in MMU firmware and connot be changed with M707/M708, but we had `MMU2_MODE_12V` with `MMU2s` and it was actually setting the stealth mode, so this should be available in the future and I'll keep this).
  - MMU Enable/Disable state
- M503 now prints MMU related data

I'm kind of ready to tidy up the code and create the PR...

And then I have some improvements that I want to make over the Prusa implementation:

- Check hotend before loading filaments to the extruder:
  As far as I see they don't check if the hotend is cold or hot before loading the filament to the extruder, I want to add checks to prevent this as in MMU2s implementation of Marlin, of course, if this is not a design decision of Prusa... I'll do my research before that..
- The "Sounds On/Off" option should be added to the menus, which should be fairly easy.

eoyilmaz, 2024-01-04:
Here is a video showing the menus and a screen capture for M503 output:

https://github.com/MarlinFirmware/Marlin/assets/1786804/7d646ab7-9e6e-4ab4-8045-4a1d7e1b2f5c

[image]

thisiskeithb, 2024-01-04:
Great progress! I'll have to dust off my MMU2 and pick up a MMU3 upgrade kit.

> Check hotend before loading filaments to the extruder:

Tying into `PREVENT_COLD_EXTRUSION` would be ideal here. Marlin also now asks if you want to continue if the nozzle temp is below `EXTRUDE_MINTEMP`.

> The "Sounds On/Off" option should be added to the menus, which should be fairly easy.

Tying into `SOUND_MENU_ITEM` / `SOUND_ON_DEFAULT` would be ideal here.

[image]

This video doesn't work in Firefox or Safari for some reason (macOS).

eoyilmaz, 2024-01-04:
> Tying into PREVENT_COLD_EXTRUSION would be ideal here. Marlin also now asks if you want to continue if the nozzle temp is below EXTRUDE_MINTEMP.

> Tying into SOUND_MENU_ITEM / SOUND_ON_DEFAULT would be ideal here.

sure sure 👍 

> This video doesn't work in Firefox or Safari for some reason (macOS).

Strange, I'm on macOS too... let me upload the original version... oh it is bigger than what GitHub accepts, here is a download link: https://www.dropbox.com/scl/fi/xu2r8mjhy4p8k2s9vyult/IMG_2856.MOV?rlkey=j1e25qas8fwxjbiugy83nhl63&dl=0

eoyilmaz, 2024-01-04:
Updated the format of MMU part of the M503 a little:

[image]

eoyilmaz, 2024-01-04:
@thisiskeithb I have a question, I'm working on enabling the SpoolJoin feature, I introduced the `AUTO` flag to `M600` and it is working fine.

The SpoolJoin feature requires checking a couple of conditions to decide to run "M600 AUTO" or not and one of the conditions is using `printJobOngoing()` at https://github.com/MarlinFirmware/Marlin/blob/994aa9f6923e2307d13badd26a15e6d57525955f/Marlin/src/MarlinCore.cpp#L342C6-L342C21

From what I understand from the function name and the content it should return `true` if the print job is on going :)

When I start the job (through OctoPrint by the way) it returns `true` until my printer homes and starts extruding (or maybe a little before) then it starts returning `false`. Is there something wrong with that function, my setup, or am I misunderstanding something?

eoyilmaz, 2024-01-05:
I was able to run the SpoolJoin, but there is an issue... At least with my setup.

The issue is that, as in my case, if the distance between the extruder gears to the nozzle tip is smaller than the distance between MMU FINDA sensor to MMU pulley gears, when the extruder pushes back the filament it will not be able push the filament back as much so that the pulley gears can grab them. I hope it makes sense... The filament stucks inside the tube between MMU FINDA and extruder, in a position nothing can pull it back.

I modified the code so that it will try consuming the filament completely before loading the next filament instead of retracting it back. I'm sure Prusa people also wanted to do this as one of the functions they are using to decide runnnig SpoolJoin is called `check_fsensor()`, but it doesn't contain anything related to filament sensor. But I'm sure their intention was to consume the filament completely and they had issues with that as a little bit of filament will be left inside the extruder, I'm hopping that it will not be a problem and I'll add a config option to Enable/Disable this behaviour (if it works at all of course).

eoyilmaz, 2024-01-05:
Interestingly, I was able to achieve what I was offering, so the filament change only occurs when the filament is completely consumed, and it is working fine... Except, after MMU loads the next filament a "Purge More" / "Continue" screen appears, I was hopping to skip this screen... Let's see...

eoyilmaz, 2024-01-05:
This is working very nicely and I found where the "Purge More" / "Continue" menu is triggered...

[image]

eoyilmaz, 2024-01-05:
The SpoolJoin is now working as intended.

I had to add two new parameters to the `resume_print` function to be passed to the `load_filament` function. Hopefully most of the places that `resume_print` is used no argument is passed to it.

I also made the "SpoolJoin Consumes All Filament" an experimental and optional feature and added the following to the `Configuration_Adv.h`:

```cpp
    /**
     * SpoolJoin Consumes All Filament -- EXPERIMENTAL
     * 
     * SpoolJoin normally triggers when FINDA sensor untriggers while printing.
     * This is the default behaviour and it doesn't consume all the filament
     * before triggering a filament change. This leaves some filament in the
     * current slot and before switching to the next slot it is unloaded.
     * 
     * Enabling this option will trigger the filament change when both FINDA
     * and Filament Runout Sensor triggers during the print and it allows the
     * filament in the current slot to be completely consumed before doing the
     * filament change. But this can cause problems as a little bit of filament
     * will be left between the extruder gears (thinking that the filament
     * sensor is triggered through the gears) and the end of the PTFE tube and
     * can cause filament load issues.
     */
    // #define MMU_SPOOL_JOIN_CONSUMES_ALL_FILAMENT
```

rerickson1, 2024-01-05:
Hey @eoyilmaz I am following your work very closely as I am interested in trying it out as soon as you have a PR ready.
I have a couple of questions I'm hoping you can help with since you have dug more than myself.
My printer is a heavily modified Anet A8 Plus. 

[image]

I have the MMU2S working pretty well.
My filament sensor at the extruder has some distance between the sensor and the gears. 

[image]

I have had to modify the MMU2S firmware and Marlin in order to feed filament reliably.
First in Marlin once the filament trips the runout sensor above the extruder I use the MMU C0 command to feed filament into the extruder gears.
To do this, I spin the extruder gears at the same speed the MMU feeds filament.

https://github.com/rerickson1/Marlin/compare/6a5f4826a7410984c53acab63b23def2d9ed408b...e61ee5eb4cdc6b5ff5739d86ec7c0c1ac0fbc7bb#diff-2e85beaab87cbd38bac78b5c7f462938214f3d7620bf669f81ff564cf291a278R698

In order for the above to work, the MMU2S firmware C0 command needed to be modified to feed more filament length. The filament sensor is greater than 38.10 mm from the extruder gears.

https://github.com/prusa3d/MM-control-01/compare/master...rerickson1:MM-control-01:master

Questions:
1. Does the new MMU3 firmware still have a C0 command? If so I can modify it to fit my needs.
2. Do you see any issues with being able to add my modifications to your new implementation? Maybe my idea for adding a setting for filament sensor distance from extruder gears is generic enough to be added to your implementation? This would help anyone trying to add MMU3 capability to any Marlin printer.
3.  Maybe this can all be accomplished in a better way now since there are many more commands for the MMU3 and better feedback?

eoyilmaz, 2024-01-05:
@rerickson1 all these should be now configurable through `Configuration_Adv.h` with MMU3 now:

```cpp
    // MMU3 Specific settings

    #define MAX_RETRIES 3 // TODO: This has the same purpose of MMU2_C0_RETRY

    // Nominal distance from the extruder gear to the nozzle tip is 87mm
    // However, some slipping may occur and we need separate distances for
    // LoadToNozzle and ToolChange.
    // - +5mm seemed good for LoadToNozzle,
    // - but too much (made blobs) for a ToolChange
    #define MMU2_LOAD_TO_NOZZLE_LENGTH 58.0F + 5.0F // 87.0F + 5.0F

    // As discussed with our PrusaSlicer profile specialist
    // - ToolChange shall not try to push filament into the very tip of the nozzle
    // to have some space for additional G-code to tune the extruded filament length
    // in the profile
    // Beware - this value is used to initialize the MMU logic layer - it will be sent to the MMU upon line up (written into its 8bit register 0x0b)
    // However - in the G-code we can get a request to set the extra load distance at runtime to something else (M708 A0xb Xsomething).
    // The printer intercepts such a call and sets its extra load distance to match the new value as well.
    #define MMU2_FILAMENT_SENSOR_POSITION 0 // mm
    #define MMU2_LOAD_DISTANCE_PAST_GEARS 5 // mm
    #define MMU2_TOOL_CHANGE_LOAD_LENGTH MMU2_FILAMENT_SENSOR_POSITION + MMU2_LOAD_DISTANCE_PAST_GEARS // mm

    #define MMU2_EXTRUDER_PTFE_LENGTH 19.0f //42.3f // mm
    #define MMU2_EXTRUDER_HEATBREAK_LENGTH  14.0f // 17.7f // mm

    #define MMU2_LOAD_TO_NOZZLE_FEED_RATE 20.0F // mm/s
    #define MMU2_UNLOAD_TO_FINDA_FEED_RATE 120.0F // mm/s

    #define MMU2_VERIFY_LOAD_TO_NOZZLE_FEED_RATE 50.0F // mm/s
    #define MMU2_VERIFY_LOAD_TO_NOZZLE_TWEAK -5.F // mm used to shorten/lenghten (negative number -> shorten) the distange of verify load to nozzle

    // The first the MMU does is initialise its axis. Meanwhile the E-motor will unload 20mm of filament in approx. 1 second.
    #define MMU2_RETRY_UNLOAD_TO_FINDA_LENGTH 80.0f // mm
    #define MMU2_RETRY_UNLOAD_TO_FINDA_FEED_RATE 80.0f // mm/s

    // After loading a new filament, the printer will extrude the filament by this distance
    // and then retract it back to the original position. This is used to check if the
    // filament sensor reading flickers or filament is jammed.
    #define MMU2_CHECK_FILAMENT_PRESENCE_EXTRUSION_LENGTH MMU2_EXTRUDER_PTFE_LENGTH + MMU2_EXTRUDER_HEATBREAK_LENGTH + MMU2_VERIFY_LOAD_TO_NOZZLE_TWEAK + MMU2_FILAMENT_SENSOR_POSITION

    #define MMU_HAS_CUTTER // Enable cutter related functionalities
    //#define MMU_FORCE_STEALTH_MODE // When defined forces the stealth mode and disables menu item
    #define MMU2_NO_TOOL 99
```

eoyilmaz, 2024-01-05:
@rerickson1 and this is very similar to my previous setup, and I just needed to adjust he MMU load lengths through the hidden menu that you access through holding the middle button while MMU starts, so that it loads pass through the filament sensor.

eoyilmaz, 2024-01-05:
@rerickson1 I just pushed my changes related to SpoolJoin feature, you can give it a try from my dev branch: https://github.com/eoyilmaz/Marlin/tree/bugfix-2.1.x_mmu3

Be aware that my branch contains `Configuration.h` and `Configuration_Adv.h` suitable for my printer and needs to be replaced as usual and I'm sure that it will not be a problem, but just in case, my pinout file for the SKR 3 EZ board has a modification to switch the fan pins. Otherwise it should work. Don't forget to copy paste the whole MMU section in `Configuration_Adv.h` file.

eoyilmaz, 2024-01-05:
Only some small fixes are left:

- Per print stats cleanup before every print
- Per print material change stat display in the menu
- Silent option

And I think that's all and I'm planning to leave these out and tidy up my branch and create the PR tonight.

eoyilmaz, 2024-01-10:
The PR is out and waiting to be reviewed, in the mean time I did a +1000 material change print (of the same Deadpool bust model, larger this time) to test the code that is in the PR right now, I had quite a bit of fails but they were all related to how I manage the filaments and the new eSun Cardboard spools are not rotating very well inside my filament dryer, eventually figured that and the print is done without any further issues:

[image]
[image]
[image]
[image]

eoyilmaz, 2024-01-22:
I'm working on the "full screen error reporting" feature and this is kind of the last item left in my table, and I'm nearly done with it. So this feature is about, showing a menu when MMU is having issues with the current action it is performing. Basically replacing the MMU hardware buttons with the printers LCD menus and encoder and it is also able to show a short text message about the error and the possible solution as seen in the video below.

https://github.com/MarlinFirmware/Marlin/assets/1786804/a128cae4-f7bf-49f8-b17e-9b787e17f865

The problems is that, in the "More Info" screen, the static items that I add to the screen is shown in a single line instead of printing each line of the info message in a new line.

I did not tidy up my branch yet so I'm going to copy&paste the code here:

```cpp
  static void show_more_info_screen() {
    if (ui.use_click())
      return ui.go_back();
    ui.clear_lcd();
    const char *msg = PrusaErrorDesc(editable.uint8);
    // LCD_ALERTMESSAGE_F(F(msg));
    const char *msgend = msg;

    START_SCREEN();
    while(pgm_read_byte(msgend) != 0) {
      // Previous row ended with a complete word, so the first character in the
      // next row is a whitespace. We can skip the whitespace on a new line.
      if (pgm_is_whitespace(msg) && ++msg == nullptr){
        // End of the message.
        break;
      }

      uint8_t linelen = (strlen_P(msg) > LCD_WIDTH) ? LCD_WIDTH : strlen_P(msg);
      const char *msgend2 = msg + linelen;
      msgend = msgend2;
      if (pgm_read_byte(msgend) != 0 && !pgm_is_whitespace(msgend) && !pgm_is_interpunction(msgend)) {
        // Splitting a word. Find the start of the current word.
        while (msgend > msg && !pgm_is_whitespace(msgend - 1)) --msgend;
        if (msgend == msg)
          // Found a single long word, which cannot be split. Just cut it.
          msgend = msgend2;
      }
      MString<LCD_WIDTH> str_buffer("");
      for (; msg < msgend; ++msg) {
        char c = char(pgm_read_byte(msg));
        if (c == '\n')
          // Abort early if '\n' is encontered.
          // This character is used to force the following words to be printed
          // on the next line.
          break;
        str_buffer.append(c);
      }
      STATIC_ITEM_F(F(str_buffer.buffer()));
    }
    END_SCREEN();
  }
```

Do you know why this happens @thisiskeithb @thinkyhead ?

eoyilmaz, 2024-01-22:
Hmmm, it might be adding 4 more items for each static item I add and visually it is giving the impression that I'm scrolling to the right and not down...

eoyilmaz, 2024-01-23:
I think it is the way this error reporting is designed, it is a state machine and it tries to work without blocking the execution, this means that I enter this function more than I expected during the message reporting...

eoyilmaz, 2024-01-23:
Ahh I now understand, the `START_SCREEN` macro is creating a `for` loop..

eoyilmaz, 2024-01-23:
Alright solved the problem by not depending on the `START_SCREEN`, `STATIC_ITEM` and `END_SCREEN` but direclty putting the characters on to the screen with `SETCURSOR` and `lcd_put_lchar()`:

[image]

thisiskeithb, 2024-01-23:
That’s a lot of text, especially for a text-based 2004 LCD like the `REPRAP_DISCOUNT_SMART_CONTROLLER`.

eoyilmaz, 2024-01-23:
> That’s a lot of text, especially for a text-based 2004 LCD like the `REPRAP_DISCOUNT_SMART_CONTROLLER`.

This is only displayed when you request more info on the matter, tbh. And there is actually a short web url also displayed as in "prusa.io/04504" to get even more info, which I didn't include yet.

eoyilmaz, 2024-01-26:
Finally, I was able to properly implement the "Fullscreen Error Messages & Actions" feature:

https://github.com/MarlinFirmware/Marlin/assets/1786804/b944ab06-a4f6-4c74-a09a-3b41b93ecf39

In the "More Info" screen the second page is displayed with a button click, that might not be very Marlin style. If required we can improve it by utilizing the encoder so that the user can scroll between pages and return to the previous screen with a button click... But I'm going to leave it to another issue ticket, and let this be reviewed and hopefully merged soon...

psharma04, 2024-05-08:
Trying to run your version on my SKR Mini E3 V3. Do you happen to know the pinout of the MMU3?
I'm trying to use this header
[image]
[image]
I have changed `MMU2_SERIAL_PORT` in `Configuration_adv.h` to 2 in an attempt to get it working, but I currently get the error "MMU not responding".

eoyilmaz, 2024-05-10:
@psharma04 you're very lucky :) I was using a SKR Mini E3 V3 before my current board. The issue that you need solve first is that the MMU board is 5V TTL and SKR Mini E3 v3, like many other 32-bit boards, is 3.3V TTL. So you need to use a Level Shifter in between. I used this for that purpose: https://www.amazon.co.uk/gp/product/B095H64XSV/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

Here is my custom board to host the level shifter: https://oshwlab.com/eoyilmaz/level-shifter

And also connect the RX port of one board to the TX port of the other and vice versa. Do not connect the RST pin, which is not needed.

thisiskeithb, 2024-05-10:
> I was using a SKR Mini E3 V3 before my current board. The issue that you need solve first is that the MMU board is 5V TTL and SKR Mini E3 v3, like many other 32-bit boards, is 3.3V TTL. So you need to use a Level Shifter in between.

If you're using the TFT port, shouldn't that work fine since it's 5V?

[image]

ellensp, 2024-05-10:
the TFT port is using PA2 (USART2_TX) /PA3 (USART2_RX) 
both of which are "FT_a" ie  they are 5 V tolerant I/O for digital inputs
but tft port will be 3.3v outputs (which is still fine, will be read as a logic 1)

microtronics, 2024-05-10:
> A level shiftier is not required

That's correct as per the datasheet and I'm using the same port on a BTT GTR.
Most STM32 pins are 5V tolerant, luckily!

eoyilmaz, 2024-05-10:
@ellensp I remember that I checked the datasheet for STM32G0B1RE when I was setting up the MMU. But, I'm not an Electrics & Electronics Engineer and I might have read it incorrectly I thought the TX pin was 4.4V tolerant and the RX was 8.4V tolerant (they give a formula based on VDD etc.). But, that might not be correct, in anyways, I'm happy that I'm using the level shifter between 3.3V TTL and 5V TTL boards, and I believe that's the general suggestion.

psharma04, 2024-05-11:
> @psharma04 you're very lucky :) I was using a SKR Mini E3 V3 before my current board. The issue that you need solve first is that the MMU board is 5V TTL and SKR Mini E3 v3, like many other 32-bit [...]

I managed to figure it out, turns out the problem was a badly crimped cable (one that I made, not from Prusa).

For future reader's reference, if you're using the MMU3 cable for the Mk3 (the one with a 5 pin header), the default order is
RX(in) TX(out) GND RESET and then the blank pinhole.

SpankyDevil, 2024-06-13:
Can you please show wiring diagram for MMU3 and SKR 3 very keen to test your great work i have the MK3S+ version

SpankyDevil, 2024-06-30:
Can i get some assistance please i am running your PR version of marlin, i have configured MMU3 Serial to be 1 which is the TFT plug on my BTT SKR 3 and used the wiring as per above post. Plus connected the 24v. I have not connected the reset.

For future reader's reference, if you're using the MMU3 cable for the Mk3 (the one with a 5 pin header), the default order is
RX(in) TX(out) GND RESET and then the blank pinhole.

I cannot comunicate with  the MMU3 as per below screen shots. The MMU3 when first powered on lights up and flashes then front lights all go off and inside you can see a blue light

[image]
[image]
[image]

SpankyDevil, 2024-07-01:
I connected the Prusa MMU3 directly to my Laptop using USB to TTL CP210x and connected via Putty, I received lots of Transmissions but nothing that I could understand as per screenshot

[image]

eoyilmaz, 2024-07-02:
@SpankyDevil I'll try to return as soon as I have time.

SpankyDevil, 2024-07-02:
Thanks,
Just an update i tried your branch for your printer and all i did was change the board model in configuration.h and tested it still came up with comunication errors but this time the status for MMU was reporting On and by running a M709 S0 i could turn it off then running M709 S1 i could turn it back on again and it reported On/Off correclty. As nothing else is connected to my board i cant find anyway to test further i did try M707 A0x04 but got no response.

I then tried the PR version and set my board model and enabled MMU3 and set Serial Port then enabled each feature as required to build and tested and i could turn MMU on/off but M709 reported MMU as Off the whole time.

I have added a 1K and 2K resistor to the MMU3 Tx line to step the voltage down to about 3.3v just incase this was causing an issue but that was done before trying your branch

Note when MMU is set to on there is a blue flashing light only and when set to off there is no ights, when i power it on the green lights come up then turn off and when set to off the green lights come on again then go off.

[image]

SpankyDevil, 2024-07-18:
Just wondering if anyone has this working with the current PR?
Would really like to get mine working.

rerickson1, 2024-07-18:
> Just wondering if anyone has this working with the current PR? Would really like to get mine working.

I have this working with my printer. I use a MKS_SGEN_L_V2 board though.

SpankyDevil, 2024-07-18:
> > Just wondering if anyone has this working with the current PR? Would really like to get mine working.
> 
> I have this working with my printer. I use a MKS_SGEN_L_V2 board though.

Would you mind confirming a few things for me please

- Are you using Baudrate 115200
- What firmware version is installed on MMU3
- Are you connected to the TFT port on your board
- Have you connected the Reset PIN
- Are you seeing any communication errors
- Are you using the MK3S cable or MK4, if using the MK3S are you using this PIN OUT

> For future reader's reference, if you're using the MMU3 cable for the Mk3 (the one with a 5 pin header), the default order is
> RX(in) TX(out) GND RESET and then the blank pinhole.

rerickson1, 2024-07-18:
* I didnt have to mess with baud rate at all. So it must be at some default.
* 3.0.2 firmware installed on my MMU2S
* No TFT, using 10 pin LCD connector
* Yes, reset pin is connected
* Flawless communication, no errors
* Using the 5 pin cable that came with my MMU2S

[image]

SpankyDevil, 2024-07-21:
> * No TFT, using 10 pin LCD connector

This version of Marlin for MMU3 requires you to have a screen enabled, so not having one physically connected but having Marlin think there is one isn't causing you any issues and running your MMU3 off of the EXP Serial.

Are you connecting through USB and manually inputting the codes to control the MMU3 which you would normally access through the Screen?

when you first power on your printer can you advise the LED status on your MMU3

rerickson1, 2024-07-22:
> This version of Marlin for MMU3 requires you to have a screen enabled, so not having one physically connected but having Marlin think there is one isn't causing you any issues and running your MMU3 [...]

I use an LCD with my printer, just not a TFT touch screen. I control the MMU3 via the LCD menus.

On boot the orange LED pulses because the main board resets it. Then LED5 flashes rapidly while the other LEDs are on and then they all turn off.

rerickson1, 2024-07-22:
The header you highlight in the picture is not used by the LCD. The two connectors on the very bottom of that picture are used by the LCD.

thisiskeithb, 2024-07-22:
> The header you highlight in the picture is not used by the LCD. The two connectors on the very bottom of that picture are used by the LCD.

I know 😄 I was clarifying/highlighting it for @SpankyDevil.

SpankyDevil, 2024-07-22:
If this is working for everyone else i must have faulty hardware, which is a real shame since the MMU3 is brand new and brought for me as a birthday gift.

I will connect the mainboard up using MMU2 method today this will eliminate the daughter board and use another cable in case either of them are at fault.

I still find it odd though that the PR branch never acknowledges the MMU3 been on but the Ender 3 branch knows when its Off/On

[5 more replies]

closed 2024-08-23 by thinkyhead