MarlinFirmware/Marlin #26526: [BUG]bed temperature is to low error

aasguard, 2023-12-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

bed temperature is to low error but bed temp is at a steady 75/75

### Bug Timeline

13/12/2023

### Actual behavior

tells me to turn printer off

### Steps to Reproduce

just start a standard print and it fails all the time 

### Version of Marlin Firmware

2.1.3e

### Printer model

voxlab aquilla x2 n32

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-13:
No configs
2.1.3e is not a Marlin version
 
Min temp errors are normally a break in the thermistor wires somewhere

aasguard, 2023-12-13:
sorry was supossed to be 2.1.2.1 i suffer with dyslexia