MarlinFirmware/Marlin #26528: [FR] Auto power loss print resume with temperature limitation

vovodroid, 2023-12-14:
### Describe the feature you want

Currently after power loss and restart printer sometimes ask confirmation for resume (M1000 S), and sometimes continue automatically (M1000) - https://github.com/search?q=repo%3AMarlinFirmware%2FMarlin+%22M1000&type=code It's hard coded for different cases (screen types?).

Confirmation approach could cause resume delay, until bed is cooled too much so model can not be further printed due adhesion loss.
Auto approach could resume printing when bed is cool and it's already too late to resume, thus causing double loss of filament.

For instance ASA on glass bed almost losses adhesion under 90-80 deg.C.

I would suggest to add resume temperature threshold parameter, with meaning to resume printing if temperature if more than threshold, and cancel (or ask) if it's already cooler.

Logic example is something like this:

```
void GcodeSuite::M1000() {

  if (recovery.valid()) {
     bool force_resume    = false;

     if (thermalManager.degBed() >= 90)
        force_resume = true;

    if (parser.seen_test('S') && !force_resume) {
```

Setting threshold to something high (or -1, or adding additional flag) will trigger current behavior. Such parameters could be define in config_adv.h, or dynamic EEPROM parameters.

If such change is welcome I could work on it.

vovodroid, 2023-12-16:
I tried to implement it in https://github.com/vovodroid/Marlin/commits/auto_resume_bed_temp/

Could it become base for PR?

thisiskeithb, 2024-01-09:
Closing since you’ve opened a PR.

PR #26649 (2024-01-10): Auto power loss print resume with temperature limitation
Closes #26528 

Currently after power loss and restart printer sometimes ask confirmation for resume (M1000 S), and sometimes continue automatically (M1000) - https://github.com/search?q=repo%3AMarlinFirmware%2FMarlin+%22M1000&type=code It's hard coded for different cases (screen types?).

Confirmation approach could cause resume delay, until bed is cooled too much so model can not be further printed due adhesion loss.
Auto approach could resume printing when bed is cool and it's already too late to resume, thus causing double loss of fi [... cut]
closed 2024-01-09 by thisiskeithb