MarlinFirmware/Marlin #26530: [BUG] SKR2 with TFT BTT_TFT35_SPI_V1_0 create white screen

gennspecial, 2023-12-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

SKR2 with TFT BTT_TFT35_SPI_V1_0 create white screen

### Bug Timeline

new. With 2.1.2.1 it was not in. when i try later Marlin-bugfix-2.1.x i found it

### Expected behavior

screen working

### Actual behavior

white screen

### Steps to Reproduce

just install the compiled firmware

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

biqu b1 se plus

### Electronics

stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/13676405/Configuration_adv.zip)

It is strange because it should have been fixed:  Fix SPI TFT for STM32F1/F4 #26052 

Sorry i am new in git hub.... maybe it is solved and i got the wrong file.... not the really last one 2.X

thisiskeithb, 2023-12-14:
Duplicate of #26504

thisiskeithb, 2023-12-14:
> It is strange because it should have been fixed: Fix SPI TFT for STM32F1/F4 https://github.com/MarlinFirmware/Marlin/pull/26052

#26052 has not been merged yet.

[image]

closed 2023-12-14 by thisiskeithb

gennspecial, 2023-12-14:
Sorry it is my first experience with marlin and github... i'm trying to understand: i have to wait that this #26052 is merged and then download the firmware again or i can already do somethings?