MarlinFirmware/Marlin #26536: [FR] Optional 'Wrap Up' Function for Early Print Termination with Final Layer Completion

0xdreadnaught, 2023-12-16:
### Is your feature request related to a problem? Please describe.

Not exactly

### Are you looking for hardware support?

No

### Describe the feature you want

There are times when printing items like dowels, containers, etc where part way through you decide "yea that's big enough for now" and want to stop the print. However you still want the top layer on it. To handle that I propose the "wrap it up" button.

When activated, the printer will complete the current layer (n). Next, it identifies the G-code commands associated with the final layer of the print. These final layer commands are put in place of the remaining layer sections, starting at n+1, Lastly the printer executes the remaining layer G-code, printing the top on the truncated print.

Given its niche application, the button should be made available as an optional feature, accessible through a settings checkbox so it doesn't clutter up the UI by default.

To take it a step further, adjustable cropping thickness would be a great addition to the settings for it. The user can specify how many layers down from the top they want to crop and slap on n+1. So if you're doing a 3 wall thick bin you would likely set it to 3.

ellensp, 2023-12-16:
Firmware does not have any concept of layers.  
It just goes where it is told to in Z
And with Z-hop it doesn't just move z in layers

0xdreadnaught, 2023-12-16:
Can it just reeval the target g-code file with the new constraints and update the remaining commands? That preserves everything you mentioned. I covered top layer thickness at the end of my request with 'adjustable crop thickness'.

Due to processing time you'ld now have to park the print head outside the current print area until the instructions have been processed and updated. otherwise you'd likely create blobs or jams.

robherc, 2023-12-16:
Unfortunately, you are asking for the MCU board to "evaluate" a file that is almost guaranteed to be bigger than not just its available RAM memory, but in many, many cases, larger than both its TOTAL RAM && its TOTAL Flash memory (i.e. even if it weren't an absolute nightmare to code that much predictive algorithm, the board pure & simple cannot hold that much data at a time).

0xdreadnaught, 2023-12-16:
Ok, thanks for the clarification. Feel free to close this because there's clearly a knowledge gap that I need to address. If I get the time to slap something together I'll reopen it. But in my head if the firmware is just streaming commands from the gcode file due to minimal resources, you'd just have to have space for 2 variabels; last layer line number and z-drop. Z-drop would just be (lastlayer−currentlayer)×layerheight. It would determine those prior to initiating the print sequence so available resources shouldn't be an issue. Now the firmware knows where to continue commands from and can easily calculate the new z-offset on the fly.

robherc, 2023-12-16:
Yes, I think you have some good ideas for what would truly be convenient...just that you don't have the coding experience with this type of thing to yet fully understand the mechanics of how the workload is shared between your slicer (or whatever g-code generator you're using) and your actual printer firmware.

If the firmware were given more.information about layer height, infill, skin layers, & etc. that generally only the slicing program is aware of, then I'd 100% back your idea here, as there have been several times in the past where I could have used such a feature. Just that, considering the vast # of options that slicers can use (check out "wire printing" in Cura for a case that would massively break any attempts here), I can't imagine any really usable way to implement this in printer firmware.