MarlinFirmware/Marlin #26537: [FR] Bed level points menu item

jzagaja, 2023-12-16:
### Is your feature request related to a problem? Please describe.

I need to add menuitem for adjusting bed level screws - 8 points x,y value so I can quickly move without computer.

### Are you looking for hardware support?

no

### Describe the feature you want

Looking for somone that can help write me such macro in menu.

EvilGremlin, 2023-12-16:
You can add this as user menu items

jzagaja, 2023-12-16:
I don't see anything.

EvilGremlin, 2023-12-16:
CUSTOM_MENU_

jzagaja, 2023-12-16:
do you have example with movement to xy point?

EvilGremlin, 2023-12-16:
https://marlinfw.org/meta/gcode/

thisiskeithb, 2023-12-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-12-16 by thisiskeithb

jzagaja, 2023-12-16:
Reprap.org forum never helps. Been there. No one helps for money?