MarlinFirmware/Marlin #26541: [BUG] Similator exits (crashes?) on Win10x64 with POWER_LOSS_RECOVERY

vovodroid, 2023-12-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Similator exits (crashes?) on Win10x64 with POWER_LOSS_RECOVERY after arriving to this point:

[image]

The only change in config https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Simulator is 

`#define POWER_LOSS_RECOVERY`

### Version of Marlin Firmware

bugfix-2.1.x Bump distribution date (2023-12-16)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mriscoc, 2024-01-29:
Did you load an SD card image?

vovodroid, 2024-05-06:
Issue is still here.

thisiskeithb, 2024-06-08:
Using the latest `bugfix-2.1.x` (320e002), I can confirm that the simulator crashes out under Windows 10 x64 (22H2 / OS Build 19045.4412) as well as macOS Monterey 12.7.5 when `POWER_LOSS_RECOVERY` is enabled.

ellensp, 2024-06-08:
crashes in Marlin/src/sd/SdVolume.h
on  uint8_t fatType() const { return fatType_; }

fatType_ is not yet initialized

thisiskeithb, 2024-06-08:
> crashes in Marlin/src/sd/SdVolume.h on uint8_t fatType() const { return fatType_; }
> 
> fatType_ is not yet initialized

Ahh. That would explain why it boots/works fine if you've generated `fs.img` & have it loaded.

closed 2024-06-16 by thinkyhead