MarlinFirmware/Marlin #26544: [BUG] MMU2S: LCD freezes when printing

rerickson1, 2023-12-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

1. MMU2S enabled in Marlin
2. When the print starts the LCD freezes and never updates
3. The print works fine and completes
4. The LCD unfreezes

### Bug Timeline

Has been present since the v2.1.2.1 release

### Expected behavior

Should be able to use the MMU2S without affecting the LCD operation.

### Version of Marlin Firmware

https://github.com/rerickson1/Marlin/tree/rde/2.1.x_2/a8%2B_custom

### Printer model

Modified Anet A8+

### Electronics

MKS SGEN L V2

### LCD/Controller

Anet A8+ LCD

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

rerickson1, 2023-12-16:
My config:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/13693871/config.zip)

My code: https://github.com/rerickson1/Marlin/tree/rde/2.1.x_2/a8%2B_custom

rerickson1, 2023-12-16:
I have tried increasing all buffers and it did not make a difference. Maybe more adjustments are needed?
```
Recv: D576 P:0 0 (0) B:3 42 (14)
Recv: D576 P:0 0 (0) B:3 36 (10)
Recv: D576 P:0 0 (0) B:3 109 (41)
```

robherc, 2023-12-16:
ummm...you're using a remix & not official marlin ... and you also haven't told what control board you're using.

rerickson1, 2023-12-17:
Control board is listed up in the description.
What do you mean by remix? I have taken vanilla marlin and applied the config settings for my board.
Without any customizations to marlin, this issue happens.

rerickson1, 2023-12-18:
I have noticed the LCD only freezes when printing is in progress and there is motion in X/Y directions.
During a tool change when there is no X/Y motion, the LCD unfreezes.
Video:
https://youtu.be/nyE4K4mlIH8
What can I try to troubleshoot this issue?

eoyilmaz, 2023-12-24:
This is a duplicate of #24813

I looked into this recently and found a couple of possible sources that can cause this, but it is still freezing... Hopefully this will fix it 😉 : #26523

rerickson1, 2024-07-22:
@thisiskeithb I would really like to help get to the bottom of this. This issue is present with this PR: https://github.com/MarlinFirmware/Marlin/pull/26635

There must be some loop starving the LCD update task. How can I enable some advanced debug? I tried increasing buffers and it didn't solve the issue, but maybe I did it wrong.

tommylux, 2024-11-10:
I'm experiencing slow performance with the LCD when printing through OctoPrint. The LCD seems to lag, moving only one notch when I turn the dial, and often appears frozen. Pressing the dial selects a different menu because the selection hasn't updated. If I wait too long, the menu often closes out. I'm wondering if the buffer is getting overloaded and if there's a way to limit the number of commands OctoPrint sends. Any suggestions would be greatly appreciated. Thanks!

Look forward to any suggestion.

Thanks

rerickson1, 2025-02-05:
@thisiskeithb I am looking into this again.  The LCD screen still freezes when printing and using MMU3. There are no pin conflicts on my board. I have ensured the UART pins I am using for the MMU3 are not used by anything else on my board schematic.
I am printing with Octoprint, not from an SD card.

I am using bug-fix-2.1.x commit 485fbf20408e19f8854e3acecb9d4e5801c8a8cd
My conifgs are here:
https://github.com/rerickson1/Marlin/blob/rde/2.1.x_mmu3-6/Marlin/Configuration.h
https://github.com/rerickson1/Marlin/blob/rde/2.1.x_mmu3-6/Marlin/Configuration_adv.h

I have adjusted buffers:
```
#define SLOWDOWN_DIVISOR 64
#define BLOCK_BUFFER_SIZE 64
#define BUFSIZE 32
#define TX_BUFFER_SIZE 128
#define RX_BUFFER_SIZE 2048
#define RX_BUFFER_MONITOR
#define MARLIN_DEV_MODE
#define BUFFER_MONITORING
```
I enabled buffer monitoring while printing:
` D576 S2`

```
Recv: D576 P:0 0 (0) B:31 13717 (58)
Recv: D576 P:0 0 (0) B:31 50 (11)
Recv: D576 P:1 0 (0) B:32 48 (10)
Recv: D576 P:0 0 (0) B:31 56 (11)
```

Based on the report, this means all 64 planner buffers are in use correct? Is that a problem? What else can I do to dig into this further?

rerickson1, 2025-02-07:
@eoyilmaz If I use M701 command in a terminal before starting the print instead of the MMU menu on the LCD to load the filament, the screen does not freeze during the print. Maybe there is an issue with the MMU LCD menus causing an issue?

eoyilmaz, 2025-02-07:
I remember that I looked into this before migrating MMU3 code, suspecting some code in MMU2(S) implementation causing it. But even after MMU3 is implemented and nothing on MMU2 code path is running this problem is still occurring. I'm not sure if MMU menus causing it as the menus are very simple and doing nothing complicated. But, I'm not experienced enough to pinpoint the issue.

rerickson1, 2025-02-07:
> I remember that I looked into this before migrating MMU3 code, suspecting some code in MMU2(S) implementation causing it. But even after MMU3 is implemented and nothing on MMU2 code path is running [...]

Appreciate the response. Whats the best way to turn off all the MMU3 serial debug messages on the terminal? I see the MMU2 had a config debug define. The MMU3 code doesn't make use of it...

RF47, 2025-03-02:
same issue witch creality v4.2.7, screen freeze/lag when printing.

SpankyDevil, 2026-08-21:
Any updates on this issue?

thinkyhead, 2026-08-21:
> The MMU3 code doesn't make use of it.

I'll have to include that in the patch for this issue. Debug logging is hard-coded but easy to wrap in that define. I'm analyzing and have some things you can try, in order to isolate the issue….

### Test 1: Is the planner genuinely full, or is the UI starved?

Start a small print that reproduces the freeze, then during the frozen period send:
```
M110 N0
```
(the newline/line-number reset) and watch whether the host still gets a response. Then poll `D576` a few times. If `P` stays pegged at the buffer maximum while X/Y is moving, the planner is genuinely saturated. This points at the LCD screen-throttling gate (we deliberately skip screen drawing when the planner is emptying faster than it is being replenished). If `P` drops but the LCD still appears frozen, then the UI is being starved by something else — a blocking wait or a re-entered idle path.

### Test 2: Does the MMU menu state carry over into the print?

Load filament through the MMU menu as you normally do, then back out of the menu to the status screen before starting the print. If the freeze no longer happens, the culprit is state left behind by the menu path rather than the print itself.

### Test 3: Is the FINDA read path implicated?

In `mmu2.cpp`, temporarily comment out the three `TERN_(HAS_PRUSA_MMU2S, check_filament())` calls inside `mmu_loop()` and rebuild. If the freeze disappears, the filament-sensor poll during idle is the likely culprit.

### Test 4: Is it re-entering the idle loop?

MMU3 guards its `mmu_loop()` against re-entry with an avoidRecursion flag, but MMU2's `mmu_loop()` does not. If you're able to enable `MARLIN_DEV_MODE`, the built-in `Marlin::idle()` call depth: message will immediately reveal whether `idle()` is being re-entered.

### Test 5: MMU2S vs MMU3

If you can switch between an MMU2S and an MMU3 (or know someone who has one) — or if you can try the same print with `MMU2_EXTRUDER_SENSOR` toggled — note whether the behavior changes. The two implementations take different idle paths, so this helps localize whether the bug lives in the shared menu layer or the device-specific polling.

The most informative single test is probably Test 1 combined with Test 4. This can tell us whether the loop is alive-but-throttled or genuinely blocked.

---

While you're checking out these behaviors I may draft a PR with other tests you can try, along with debug logging being made switchable.

closed 2026-10-27 by thinkyhead