MarlinFirmware/Marlin #26548: [FR] Add solid state relay based bed heater option. PID controlled. 1Hz PWM frequency.

TBAMax, 2023-12-17:
### Is your feature request related to a problem? Please describe.

Solid state relay(SSR) based heater.
Brief:

1. SSR based heaters work directly from mains voltage 50Hz or 60Hz
2. SSRs usually have Zero cross switching, his means relay is swiched on or off only at the moment when the mains voltage passes zero. This means swiching can be done at 100Hz maximum for 50 Hz mains voltage, once per half period. Minimum single switched on period is one half of the mains sine wave.
3. To achieve the smooth PWM regulation in this case the PWM frequency must be very low. 1Hz would be a good PWM period to use as this gives nice 100 levels in the full range of regulation.
4. From the controller side it do not need to be synchronized to the mains frequency. It is enough to drive the SSR unsynchronized, it still behaves well (tested in practice).
5. Need for PWM regulation arises from the possibly big power of the heating element. Can not use just BANG-BANG system because it would melt the bed during the long on period (for example when using 2000W heater element salvaged from old oven).
6. Currently there is no support for such low PWM frequency and also no support for separate bed PWM frequency.
7. Concept is tested on other platforms and working fine. Separate Arduino Nano based bed heater controller, or Smoothieware with PWM set to 1Hz and PID tuned. 

### Are you looking for hardware support?

AC powered heated bed heaters. Can be direct from mains, or via transformer (for extra safety).

### Describe the feature you want

PID control with 1Hz PWM @ 1Hz for heated bed.

EvilGremlin, 2023-12-17:
What's wrong with default ~7Hz PWM? It works fine with zero-crossing. Also optotriacs only turn on at zero, but turn off happens immediately. Though charge usually carry triac open to the end of half-period. Big heaters are too inert to bother with fine PWM control, there in no noticeable difference between 10 and 100 steps.

TBAMax, 2023-12-18:
> What's wrong with default ~7Hz PWM? It works fine with zero-crossing. Also optotriacs only turn on at zero, but turn off happens immediately. Though charge usually carry triac open to the end of [...]

Would not want to compromise the hotend temperature regulation because of the heated bed requirements. For hotend the ~7Hz seems way too low, it might not respond adequately for variations in the extrusion rate any more. Wait... what is the default frequency for hotend heaters at the moment... need to check that? So many things to research... I make the list for now about the topics:

1. Is it true that currently hotends and beds an all other heater PWMs only can run all at the same PWM frequency? That there is no way to configure them separately? And currently configuring heated bed PWM to ~7Hz would cause hotends PWM to run at the same frequency?
2. What is the default PWM frequency anyway at the moment?... just need to confirm that, as it came into the discussion. Is it really 7Hz for all PWMs in Marlin currently? I had impressinon it was a bit higher by default... 
3. Why it is so hard to implement with the current codebase? Different platforms having different resources.

EvilGremlin, 2023-12-18:
All heaters PWM was always 7Hz

robherc, 2023-12-18:
I don't think MPC changes the PWM base frequency any...just predicts what changes are about to be needed, rather than PID reacting 2-5 seconds later, once the thermistor registers the change.
However, tbf, I don't see that there's likely any inherent need to have a 7hz base-freq. for any non-induction hotends (pretty sure that one uses a separate controller anywise), so maybe we could change the default PWM freq. to 3hz, or just make it user-configurable with a `#define PWM_HEATING_FREQUENCY 7` line in configuration_adv.h ?
(By my understanding, a mains-powered bed heater is likely going to change the bed temp faster than a 40-50w cartridge-heater will change the hotend temp anywise).

EvilGremlin, 2023-12-18:
Yep, it doesn't.
Anyway, to propose such changes you must provide real evidence of it's necessity. For now, everything you said is at superstition level.

EmperorArthur, 2023-12-22:
You'll have to forgive me as someone new to this project, and who hasn't recently done anything with mains dimming.  This whole question just seems odd to me.

@TBAMax Re. Number 5.  Never have a single piece of software be your only safety mechanism.  Infinitely more so if it's not deliberately designed safety critical software.  Mains thermostat switches are cheap and will save you.  The only reason you don't see them on everything is PTC heaters are self-limiting and ovens are specifically designed with the maximum heater temperature in mind.

@EvilGremlin PWMing a tirac without synchronizing to the zero crossing seems weird to me.  Doubly so if you're talking an SSR with zero crossing detection.  At that point, the minimum on time is 1/2 wave, but it only triggers if the signal is high at the zero crossing. Which for a PWM is essentially random unless you filter it and are dealing with if the voltage is high enough to trigger the SSR.

tl;dr:  A Solid State Relay should probably be treated as a relay, not a dimmer.

EvilGremlin, 2023-12-22:
>PWMing a tirac without synchronizing to the zero crossing seems weird 

It would be, were our heaters less inert

EmperorArthur, 2023-12-23:
> It would be, were our heaters less inert

Not sure what you mean in this context.

robherc, 2023-12-23:
My guess is that he means they're pretty stable & rather slow-acting ... so any offsets cause by non-synchronization between the PWM frequency & the mains 0-crossings simply gets neutralized by being averaged-out by the thermal inertia of the system.

InsanityAutomation, 2024-04-07:
There is a config option not explicitly spelled out that should be able to achieve what you want - 

```
#if ENABLED(SLOW_PWM_HEATERS) && !defined(MIN_STATE_TIME)
  #define MIN_STATE_TIME 16 // MIN_STATE_TIME * 65.5 = time in milliseconds
#endif
```
If you set the MIN_STATE_TIME along with enabling slow_pwm_heaters you should be able to get the rate youre looking for.

As many platforms share PWM timers for fans and heaters, it most often isnt technically feasible to make it bed only unfortunately.