MarlinFirmware/Marlin #26552: [BUG] LED control in encoder.cpp - invalid narrowing conversion

classicrocker883, 2023-12-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

`invalid narrowing conversion from "int" to "unsigned long"`

[image]

I was messing around and saw `LED_Action()` commented out and wondered why
```y
      #if PIN_EXISTS(LCD_LED)
        //LED_Action();
      #endif
```

so I enabled that pin, and stumbled upon this error.  

one possible fix is  
```diff
-    struct { uint8_t g, r, b; } led_data[LED_NUM];
     for (uint8_t i = 0; i < LED_NUM; i++) {
       switch (RGB_Scale) {
         case RGB_SCALE_R10_G7_B5:
-           led_data[i] = { luminance * 7/10, luminance * 10/10, luminance * 5/10 };

+    struct { uint32_t g, r, b; } led_data[LED_NUM];
     for (uint8_t i = 0; i < LED_NUM; i++) {
       switch (RGB_Scale) {
         case RGB_SCALE_R10_G7_B5:
+          led_data[i] = { luminance * (u_int)7/10, luminance * (u_int)10/10, luminance * (u_int)5/10 };
```

### Bug Timeline

could be older, only noticed it now

### Expected behavior

should not give error for invalid conversion

### Actual behavior

gives error for invalid conversion

### Steps to Reproduce

1. inside **pins_(YOUR_BOARD).h** -- i.e. **Marlin\src\pins\stm32f1\pins_CREALITY_V4.h**
2. add `#define LCD_LED_PIN  EXP3_02_PIN` -- or any pin
3. go to **encoder.cpp** and you will see the error

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Aquila

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-20:
```DIFF
- struct { uint8_t g, r, b; } led_data[LED_NUM];
 +    struct { uint32_t g, r, b; } led_data[LED_NUM];
 
```  

So you want to use 4 bytes for each of R,G and B that can only be 0-255 and fits with in a single byte??

robherc, 2023-12-20:
That error is saying you're converting from a signed 1+byte (depending on arch) type to an unsigned byte type. I don't think converting to a `uint128_t` would get rid of the warning anywise, as it still couldn't store -3. ...might need to find where the variable oor data was declared as a signed type to fix this.

classicrocker883, 2023-12-21:
> ```diff
> - struct { uint8_t g, r, b; } led_data[LED_NUM];
> +    struct { uint32_t g, r, b; } led_data[LED_NUM];
> ```
> 
> So you want to use 4 bytes for each of R,G and B that can only be 0-255 and fits with in a single byte??

well no, i tried different things and that was the first which cleared the error. if it were me making the choice then definitely not, but something else.

its just an example of how to think, not what to think.

thisiskeithb, 2024-03-28:
Closing since you’ve opened a PR.

closed 2024-03-28 by thisiskeithb