MarlinFirmware/Marlin #26559: [BUG] M48 not completing

Tannoo, 2023-12-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Execute M48 homes all axi and stops with saying the probe is out of bounds.

### Bug Timeline

Noticed when setting up and testing a BLTouch for the first time.

### Expected behavior

It needs to return the probe to a probe-able point and continue.

### Actual behavior

Execute M48 homes all axis and stops with saying the probe is out of bounds.

### Steps to Reproduce

Probe sitting at X100 Y100 with steppers disabled or on power up.
Execute M48 homes all axi and stops with saying the probe is out of bounds.
The probe was NOT out of bounds when M48 was executed... it PUT it out of bounds.

### Version of Marlin Firmware

Marlin: bugfix-2.1.x (Dec 21 2023 12:52:56)

### Printer model

Anycubic Kobra Max

### Electronics

BTT SKR 3 EZ

### LCD/Controller

BTT TFT35 E3 V3.0.1

### Other add-ons

BLTouch

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://drive.google.com/file/d/1WKsWA-tGgmrZOzI4BZ-Ge8SZguiA-GON/view?usp=drive_link

thisiskeithb, 2023-12-21:
Your link is password protected (which is why we ask you to attach your config here as a .zip file), but you likely forget to enable `Z_SAFE_HOMING`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Tannoo, 2023-12-21:
Oh. I tried to upload the zip file and it didn't seem to take it.

Z safe homing is not enabled because I have a z min limit that is not using the probe.

Looking closer at this machine, it looks like the nozzle is still 1mm or so away from the bed when the optical sensor limit is triggered. I have to verify that before I enable z_safe_homing.

Thank you.

Tannoo, 2023-12-21:
Sorry for the inconvenience.

Tannoo, 2023-12-31:
This is still an issue. If the machine is not homed and M48 is issued, Z min limit is ignored.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/13800210/Configs.zip)

thisiskeithb, 2023-12-31:
I wouldn't call this a bug. `M48` requires that your machine be homed at least once after power up:

https://github.com/MarlinFirmware/Marlin/blob/5d1ede08aa8a045463d4d33333fd90abead2871e/Marlin/src/gcode/calibrate/M48.cpp#L56

https://github.com/MarlinFirmware/Marlin/blob/5d1ede08aa8a045463d4d33333fd90abead2871e/Marlin/src/gcode/calibrate/M48.cpp#L61-L61

_Note: `homing_needed_error()` is the check that your machine has homed at least once after power up, not that homing is required / axis are untrusted._

Once your machine is homed at least once, `M48` will probe at the current position without homing.

Confirmed here on a Prusa Bear with PINDA & Biqu Hurakan with MicroProbe:

1. Home your machine
2. Move to the position you want to probe
3. Send `M48` or run "M48 Probe Test" from the LCD. The probe repeatability test will run.

You can also disable steppers / let them time out and send `M48` or run "M48 Probe Test" from the LCD again and it'll probe in probe in place without homing.

Another option is to run `M48 X<pos> Y<pos>` to specify a probe location.

Our [`M48 - Probe Repeatability`](https://marlinfw.org/docs/gcode/M048.html) page needs to be updated with this information.

ellensp, 2023-12-31:
just attempting M48 when  not homed and it  stops Z endstop working... this is not a desirable outcome

thisiskeithb, 2023-12-31:
> just attempting M48 when not homed and it stops Z endstop working... this is not a desirable outcome

If you've homed at least once and run `M48`, Z will home until the probe triggers. If you run `M48` in a spot that the probe cannot sense the bed, well, ...don't do that 😄

Tannoo, 2023-12-31:
That is not the issue.

The probe is not the Z min.

Tannoo, 2023-12-31:
Attempting M48 when not homed, causes M48 to initiate G28. X and Y home and Z attempts to home, but ignores the Z endstop.

Tannoo, 2023-12-31:
Z_SAFE_HOMING is enabled, but that is not the issue. Again, the probe is NOT the Z endstop.

Tannoo, 2023-12-31:
Because the probe is not the Z endstop, the BLTouch probe is never deployed when homing Z.

Tannoo, 2023-12-31:
So, this is more of a G28 issue. M48 initiates G28, but because I do not have the probe setup as the Z endstop, it is ignoring the actual Z endstop?

This does not happen any other time I issue a G28. The endstop works just fine. The probe works just fine also. I got UBL to probe all points on a 7x7 grid. Not once did it crash.

Again... M48 works just fine, if the machine is homed first. And yes, it will probe where it sits as long as it is homed first.

Tannoo, 2023-12-31:
Sorry... the issue discription is incorrect. I can change it or I can start a new issue.

closed 2023-12-31 by Tannoo