MarlinFirmware/Marlin #26566: [BUG]  'DISPLAY_SLEEP_MINUTES' was not declared in this scope

avolkov, 2023-12-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to compile marlin for mks_robin_nano_v1v2 I get the following error 

```
Compiling .pio/build/mks_robin_nano_v1v2/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/HAL/shared/esp_wifi.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/HAL/shared/servo.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/MarlinCore.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/core/serial.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/core/utility.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/ammeter.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/bdl/bdl.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/bedlevel.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o
In file included from Marlin/src/MarlinCore.cpp:64:
Marlin/src/lcd/marlinui.h:289:56: error: 'DISPLAY_SLEEP_MINUTES' was not declared in this scope
  289 |       static constexpr uint8_t sleep_timeout_minutes = DISPLAY_SLEEP_MINUTES;
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/mks_robin_nano_v1v2/src/src/MarlinCore.cpp.o] Error 1
In file included from Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:33:
Marlin/src/feature/bedlevel/ubl/../../../lcd/marlinui.h:289:56: error: 'DISPLAY_SLEEP_MINUTES' was not declared in this scope
  289 |       static constexpr uint8_t sleep_timeout_minutes = DISPLAY_SLEEP_MINUTES;
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o] Error 1
```

The error doesn't seem to be affected whenever DISPLAY_SLEEP_MINUTES defined in Configuration_adv.h or not

### Bug Timeline

Durrently reproducible, first started around 775c6bb20e

### Expected behavior

Source code compiles

### Actual behavior

Compile error for UBL

```
Compiling .pio/build/mks_robin_nano_v1v2/src/src/core/utility.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/ammeter.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/bdl/bdl.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/bedlevel.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl.cpp.o
Compiling .pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o
In file included from Marlin/src/MarlinCore.cpp:64:
Marlin/src/lcd/marlinui.h:289:56: error: 'DISPLAY_SLEEP_MINUTES' was not declared in this scope
  289 |       static constexpr uint8_t sleep_timeout_minutes = DISPLAY_SLEEP_MINUTES;
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/mks_robin_nano_v1v2/src/src/MarlinCore.cpp.o] Error 1
In file included from Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:33:
Marlin/src/feature/bedlevel/ubl/../../../lcd/marlinui.h:289:56: error: 'DISPLAY_SLEEP_MINUTES' was not declared in this scope
  289 |       static constexpr uint8_t sleep_timeout_minutes = DISPLAY_SLEEP_MINUTES;
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/mks_robin_nano_v1v2/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o] Error 1
```

### Steps to Reproduce

Copy Configuration.h, Configuration_adv.h and platformio.ini from the zip file. Run `pio run`

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

kp3s

### Electronics

mks_robin_nano_v1v2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[marlin-2.1.x-bug.zip](https://github.com/MarlinFirmware/Marlin/files/13760320/marlin-2.1.x-bug.zip)

ellensp, 2023-12-24:
TFT_COLOR_UI doesn't set HAS_MARLINUI_U8GLIB, so that section in Configuration_adv.h that sets DISPLAY_SLEEP_MINUTES is skipped.

a simple fix might be to just edit Configuration_adv.h 

#if HAS_MARLINUI_U8GLIB => #if HAS_MARLINUI_U8GLIB || HAS_GRAPHICAL_TFT

thisiskeithb, 2023-12-24:
> a simple fix might be to just edit Configuration_adv.h

Or look for `EDITABLE_DISPLAY_TIMEOUT` / `HAS_DISPLAY_SLEEP` instead.

InsanityAutomation, 2024-04-07:
Closing this as #26747 snuck in the resolution.

closed 2024-04-07 by InsanityAutomation