MarlinFirmware/Marlin #26571: [BUG] YHCB2004 - undefined reference to `operator delete(void*, unsigned int)'

NoahJaehnert, 2023-12-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Pulled the latest version of Marlin Bugfix-2.1.x ([bb557e5](https://github.com/MarlinFirmware/Marlin/commit/bb557e5195a6177bc7386fce0176274aede60b64)). 

Upon compiling, I get the following error: 
```
In file included from .pio/libdeps/mega2560ext/LiquidCrystal/LiquidCrystal_SR1W.cpp:48:0:
.pio/libdeps/mega2560ext/LiquidCrystal/LiquidCrystal_SR1W.h:168:1: warning: multi-line comment [-Wcomment]
 //             |    |      0.1uF            |     \
 ^
/var/folders/3h/zglp2t_141q_7sdtq0d0lf700000gn/T//ccN293xt.ltrans4.ltrans.o: In function `global destructors keyed to 65535_1_HAL.cpp.o.21605':
<artificial>:(.text.exit+0x28): undefined reference to `operator delete(void*, unsigned int)'
collect2: error: ld returned 1 exit status
```

Per a conversation in #support-chat on Discord, Dust confirmed "enabling #define YHCB2004 on stock ramps configs is sufficient to triggger it" and that the bug was introduced here: https://github.com/MarlinFirmware/Marlin/commit/ec7ab5a277a0210e1349f9e8608c372e40fdb6e6

Also, per Dust: "that library does not like the changes in avr.ini (reverting just changes to avr.ini and it builds again)"

### Bug Timeline

New bug

### Expected behavior

Expect to be able to successfully compile Marlin

### Actual behavior

Unable to successfully compile Marlin

### Steps to Reproduce

1. Pull ([bb557e5](https://github.com/MarlinFirmware/Marlin/commit/bb557e5195a6177bc7386fce0176274aede60b64)) version of Marlin Bugfix-2.1.x 
2. Enable/Define YHCB2004 (#define YHCB2004)
3. Compile Marlin
4. Receive error as stated above

### Version of Marlin Firmware

Marlin bugfix-2.1.x 

### Printer model

Geeetech A10M

### Electronics

Geeetech A10M GT2560_V4.1B board with TMC2208 stepper drivers

### LCD/Controller

Geeetech A10M Default (YHCB2004)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-25:
YHCB2004  uses these two external libraries  red-scorp/LiquidCrystal_AIP31068@^1.0.4, red-scorp/SoftSPIB@^1.1.1

ellensp, 2023-12-26:
also from  https://gcc.gnu.org/onlinedocs/gcc/C-Dialect-Options.html "The name ‘gnu++1z’ is deprecated"

```
‘gnu++17’
‘gnu++1z’
GNU dialect of -std=c++17. This is the default for C++ code. The name ‘gnu++1z’ is deprecated.
```

ellensp, 2024-05-08:
@NoahJaehnert Can you test a potential fix for me? I know it builds

update features.ini 
replace
`YHCB2004                               = red-scorp/LiquidCrystal_AIP31068@^1.0.4, red-scorp/SoftSPIB@^1.1.1
`

with
`YHCB2004                               = LiquidCrystal_AIP31068=https://github.com/ellensp/LiquidCrystal_AIP31068/archive/refs/heads/update-for-marlin.zip, red-scorp/SoftSPIB@^1.1.1
`

This forces marlin to use the library from my repo with the updates
Jist for Information: The important change being
https://github.com/red-scorp/LiquidCrystal_AIP31068/commit/3fc43b70d34dba5a5aae39024906e1929a0ee3a2

closed 2024-07-05 by thinkyhead