MarlinFirmware/Marlin #26575: Problem/Errors integrating MAX31865 with Ultimaker Original (UMO) for Heated Bed:

SimPitTech, 2023-12-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Background:
The goal was to integrate the MAX31865 RTD-to-Digital converter module with the Ultimaker Original (UMO) 3D printer to manage the heated bed temperature. The MAX31865 is used for controlling heated bed only and I've run into variety of errors (more about it in the actual steps).

Initial Setup:
The setup involved configuring the Marlin firmware to recognize and correctly use the MAX31865 module in Configuration.h and Configuration_adv.h 

Troubleshooting Steps:

    Configuration Files Review:
        Ensured that TEMP_SENSOR_BED was set to -5 for the MAX31865 in Configuration.h.
        Configured specific settings related to the MAX31865 in Configuration_adv.h, including settings for the number of sensor wires, filtering options, and SPI settings.

    Compilation Errors Analysis:
        Addressed various compilation errors related to the BED_MAXTEMP and BED_MAX_TARGET macros. The firmware failed to compile with these macros in their default configuration as those terms were not recognized. When set to other sensor macros work correctly. 

    Hardcoding Values:
        As a workaround, hardcoded the values of BED_MAXTEMP and BED_MINTEMP in various parts of the firmware to bypass errors related to these macros. This approach successfully resolved the compilation issues but brought in new ones.

    LCD Menu Adjustments:
        Commented out related menu options in the LCD interface that were affected by the hardcoded temperature values to prevent conflicts during user interactions.

    Further Debugging:
        Despite hardcoding values, additional issues were encountered, particularly in the menu_configuration.cpp file, related to menu items for bed temperature settings.

I've tried few older marlin versions as well and the problem remains the same. Is there a bug that needs to be fixing or is the problem between the chair and the keyboard? 

Thanks for help.

### Steps to Reproduce

#define TEMP_SENSOR_BED -5 and compile. (i've used Visual Studio Code and Autobuild Marlin) 

### Version of Marlin Firmware

2.1.12

### Printer model

Ultimaker Original

### Electronics

umo1.5.7

### LCD/Controller

original marlin 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13774030/Marlin.zip)

thisiskeithb, 2023-12-26:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

You sure? `MAX31865` support for the bed was added in the following PR and added to `bugfix-2.1.x` last month: 

- https://github.com/MarlinFirmware/Marlin/pull/26441

SimPitTech, 2023-12-27:
Thanks - I'll try again with bugfix 2.1 to eliminate the possibility the problem is between chair and keyboard.

SimPitTech, 2023-12-27:
> > Did you test the latest bugfix-2.1.x code?
> > Yes, and the problem still exists.
> 
> You sure? `MAX31865` support for the bed was added in the following PR and added to `bugfix-2.1.x` last month:
> 
>     * [Add max31865 support for heated bed #26441](https://github.com/MarlinFirmware/Marlin/pull/26441)

I still get errors:
In file included from Marlin\src\MarlinCore.cpp:207:0:
Marlin\src\feature/hotend_idle.h: In member function 'void<unnamed struct>::set_defaults()':
Marlin\src\feature/hotend_idle.h:35:5: error: 'bed_target' was not declared in this scope
     bed_target    = HOTEND_IDLE_BED_TARGET;
     ^~~~~~~~~~
Marlin\src\feature/hotend_idle.h:35:5: note: suggested alternative: 'nozzle_target'
     bed_target    = HOTEND_IDLE_BED_TARGET;
     ^~~~~~~~~~
     nozzle_target
In file included from Marlin\src\feature\hotend_idle.cpp:32:0:
Marlin\src\feature\hotend_idle.h: In member function 'void<unnamed struct>::set_defaults()':
Marlin\src\feature\hotend_idle.h:35:5: error: 'bed_target' was not declared in this scope
     bed_target    = HOTEND_IDLE_BED_TARGET;
     ^~~~~~~~~~
Marlin\src\feature\hotend_idle.h:35:5: note: suggested alternative: 'nozzle_target'
     bed_target    = HOTEND_IDLE_BED_TARGET;
     ^~~~~~~~~~
     nozzle_target
*** [.pio\build\mega2560\src\src\MarlinCore.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\feature\hotend_idle.cpp.o] Error 1

ellensp, 2023-12-27:
you have not set a bed thermistor type
you have
#define TEMP_SENSOR_BED 0

you need #define TEMP_SENSOR_BED -5

ellensp, 2023-12-27:
Code is not able to have 1 MAX31865 as the bed, that is a scenario not catered for

ellensp, 2023-12-27:
I got it to build on bugfix 2.1.x, I cannot test it

Using these configs (not you need to change the pins I used lines 74-77 to the pins you used)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13775455/Configuration.zip)

New Marlin/src/module/temperature.h and Marlin/src/module/temperature.cpp
[temperature.zip](https://github.com/MarlinFirmware/Marlin/files/13775466/temperature.zip)

New Marlin/src/inc/Conditionals_adv.h
[Conditionals_adv.zip](https://github.com/MarlinFirmware/Marlin/files/13775474/Conditionals_adv.zip)

SimPitTech, 2023-12-27:
Thanks, when testing I had it set to -5. I'll give it a go with the files and report in a moment.

SimPitTech, 2023-12-27:
> I got it to build on bugfix 2.1.x, I cannot test it beyond compiling
> 
> Using these configs (note you need to change the MAX31865 pins, lines 74-77, to the pins you used) [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13775455/Configuration.zip)
> 
> New Marlin/src/module/temperature.h and Marlin/src/module/temperature.cpp [temperature.zip](https://github.com/MarlinFirmware/Marlin/files/13775466/temperature.zip)
> 
> New Marlin/src/inc/Conditionals_adv.h [Conditionals_adv.zip](https://github.com/MarlinFirmware/Marlin/files/13775474/Conditionals_adv.zip)

Big thanks - it does compile now and the marlin loads up ok but I'm getting error with min bed temp on 3d printer which I think is wiring issue now. I've setup the pins the way I have it wired. 
#define MOTHERBOARD BOARD_ULTIMAKER
#define TEMP_BED_CS_PIN 10
#define TEMP_BED_MISO_PIN  12
#define TEMP_BED_SCK_PIN   13
#define TEMP_BED_MOSI_PIN  11
[image]

but I think i need to solder the 3 pads for 2 wires connection (just noticed I haven't done it). I'll fix it and report hopefully soon the results

SimPitTech, 2023-12-27:
Still min bed temp error. I've soldered the pads, checked power and wiring again and I'm still getting min bed temp error

ellensp, 2023-12-28:
with a two wire, the probe is plugged into both two pin connectors, one wire in the left block of two, one wire in the right block of two. 

like so on the left.  (the right is fr a 3.3v system, so do not copy that)  

[image]

ellensp, 2023-12-28:
pins you have selected are in conflict with
#define SERVO0_PIN                        11
#define PS_ON_PIN                          12
#define LED_PIN                               13

the led pin is controlled by marlin
as the PS_ON_PIN is set to an input 
the servo pin may also be set off, 
make sure these are updated to -1 to disable them

dbuezas, 2024-06-05:
I've been using not one but two of these for both the bed and hotend of an UM2 without issues for 6 months. I assume this issue is either a wiring problem or a broken max chip.

I.e I think it can be closed

closed 2024-06-05 by thisiskeithb