MarlinFirmware/Marlin #26577: [BUG] LAtest Bugfix Fails to compile for Robin Nano v1.2 )

ahmetcemturan, 2023-12-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to compile for the Robin NAno 1.2 I get the following message:

### Bug Timeline

It is existent in the latest bugfix

### Expected behavior

Compile correctly

### Actual behavior

Fails and gives following error:  
In file included from Marlin\src\lcd/dogm/u8g_fontutf8.h:11,
                 from Marlin\src\lcd/lcdprint.h:53,
                 from Marlin\src\lcd/marlinui.h:74,
                 from Marlin\src\MarlinCore.cpp:64:
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:944:110: error: 'u8g_dev_ssd1309_128x64_f_sw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_sw_spi'?
  944 |   void init(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_sw_spi, sck, mosi, cs, a0, reset); }
      |                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                              u8g_dev_ssd1309_128x64_sw_spi
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:945:83: error: 'u8g_dev_ssd1309_128x64_f_hw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_hw_spi'?
  945 |   void init(uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_hw_spi, cs, a0, reset); }
      |                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                   u8g_dev_ssd1309_128x64_hw_spi
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:946:65: error: 'u8g_dev_ssd1309_128x64_f_i2c' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_i2c'?
  946 |   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_i2c, options); }
      |                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                 u8g_dev_ssd1309_128x64_i2c
*** [.pio\build\mks_robin_nano_v1v2\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\feature\../lcd/dogm/u8g_fontutf8.h:11,
                 from Marlin\src\feature\../lcd/lcdprint.h:53,
                 from Marlin\src\feature\../lcd/marlinui.h:74,
                 from Marlin\src\feature\cancel_object.cpp:29:
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:944:110: error: 'u8g_dev_ssd1309_128x64_f_sw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_sw_spi'?
  944 |   void init(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_sw_spi, sck, mosi, cs, a0, reset); }
      |                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                              u8g_dev_ssd1309_128x64_sw_spi
In file included from Marlin\src\feature\../lcd/dogm/u8g_fontutf8.h:11,
                 from Marlin\src\feature\../lcd/lcdprint.h:53,
                 from Marlin\src\feature\../lcd/marlinui.h:74,
                 from Marlin\src\feature\hotend_idle.cpp:38:
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:944:110: error: 'u8g_dev_ssd1309_128x64_f_sw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_sw_spi'?
  944 |   void init(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_sw_spi, sck, mosi, cs, a0, reset); }
      |                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                              u8g_dev_ssd1309_128x64_sw_spi
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:945:83: error: 'u8g_dev_ssd1309_128x64_f_hw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_hw_spi'? 
  945 |   void init(uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_hw_spi, cs, a0, reset); }
      |                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                   u8g_dev_ssd1309_128x64_hw_spi
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:945:83: error: 'u8g_dev_ssd1309_128x64_f_hw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_hw_spi'?
  945 |   void init(uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_hw_spi, cs, a0, reset); }
      |                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                   u8g_dev_ssd1309_128x64_hw_spi
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t)':
.pio\libdeps\mks_robin_nano_v1v2\U8glib-HAL\src/U8glib-HAL.h:946:65: error: 'u8g_dev_ssd1309_128x64_f_i2c' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_i2c'?       
  946 |   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_i2c, options); }
[... 62 lines cut ...]
### Electronics

Robin Nano 1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13778776/Marlin.zip)

I tried to compile by commenting out the whole section that throws the errors and it compiled but then the printer got stuck at "updating 100%" .  After reboot it hangs at "booting".

ellensp, 2023-12-27:
recent changes to U8glib-HAL have broken things https://github.com/MarlinFirmware/U8glib-HAL/commit/c1ec1a465274669b4e47ed462d3a09d62cdc9b75

ellensp, 2023-12-27:
fixing your max temp to #define HEATER_0_MAXTEMP 260 and forcing marlin to use older U8glib-HAL

Change feature.ini line
HAS_MARLINUI_U8GLIB                    = marlinfirmware/U8glib-HAL@~0.5.2
to
HAS_MARLINUI_U8GLIB                    = marlinfirmware/U8glib-HAL@0.5.2

Will allow you to build right now, till this issue is fixed properly

ellensp, 2023-12-27:
Can also be fixed by editing .pio/libdeps/mks_robin_nano_v1v2/U8glib-HAL/src/clib/u8g.h

And adding the block 

```
/* OLED 128x64 Display with SSD1309 Controller - Fast Single Page */
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_sw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_hw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_i2c;

```

after the block

```
/* OLED 128x64 Display with SSD1309 Controller */
extern u8g_dev_t u8g_dev_ssd1309_128x64_sw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_hw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_i2c;

```

ellensp, 2023-12-27:
Issue opened on U8glib-HAL https://github.com/MarlinFirmware/U8glib-HAL/issues/32

ahmetcemturan, 2023-12-27:
It compiles but screen freezes at: Updating 100%

ahmetcemturan, 2023-12-27:
The same happens with this solution stuck at 100% :(

thisiskeithb, 2023-12-27:
> The same happens with this solution stuck at 100% :(

In addition to the `HAS_MARLINUI_U8GLIB = marlinfirmware/U8glib-HAL@0.5.2` change, you may also need to roll back your fork of `bugfix-2.1.x` to c485f513d70acc838b7a1554c955fdaf5ef3df7f which is right before "🧑‍💻 Move U8G defines to HALs" / bd36644d85bef0ff02f701203365ce16c5db3dc0

thinkyhead, 2023-12-27:
If it is confirmed that rolling back that commit fixes an issue then I need another set of eyes on bd36644d85. I don't see how that commit changes anything that would affect a Robin Nano 1.2 (or indeed anything at all, since it just moves the u8g assignments to the HALs but does not change them).

thinkyhead, 2023-12-27:
Once U8G 0.5.4 propagates to the PIO registry the build issue will be fixed and this can be closed.

> It compiles but screen freezes at: Updating 100%

That may or may not be related to the U8G changes. When was the last time you re-flashed Marlin before today?

thisiskeithb, 2023-12-27:
> Once U8G 0.5.4 propagates to the PIO registry the build issue will be fixed and this can be closed.

For completeness, I forced Marlin to use `0.5.4` with `HAS_MARLINUI_U8GLIB = U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/refs/tags/0.5.4.zip` with current `bugfix-2.1.x` and tested changes on a BTT Mini 12864 V2 (with NeoPixels) and it works fine.

Not sure about ahmetcemturan's issue though.

ahmetcemturan, 2023-12-28:
Rolling Back still does not work for me.  It gets stuck at 100% and after
Reboot "Booting...."
I want to mention that my nano 1.2 is the one with the onboard drivers..
That's what I got after frying the other one :(
Maybe it makes a difference..

ahmetcemturan, 2023-12-28:
To clarify: I am not using github for tracking.  So I downloaded from
https://github.com/MarlinFirmware/Marlin/tree/c485f513d70acc838b7a1554c955fdaf5ef3df7f
Adapted my Configuration files and I changed features.ini. That's all..

ellensp, 2023-12-28:
On board drivers?  url to the board? as thats not standard and I wonder if other things have been changed,

ahmetcemturan, 2023-12-28:
I think it is called nano s.   I will open it up and look at the board when I get home-- Ahmet Cem Turan+90 555 455 5444

thisiskeithb, 2023-12-28:
Have you successfully ran a prior version of Marlin on this board or is it brand new?

ahmetcemturan, 2023-12-28:
Yes of course..  I am reverting to it every time.
#define STRING_DISTRIBUTION_DATE "2023-10-12"

ahmetcemturan, 2023-12-28:
Yes of course..  I am reverting to it every time.
#define STRING_DISTRIBUTION_DATE "2023-10-12"
and the board is marked as: MKS Robin Nano-S V1.3
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X
--
Ahmet Cem Turan
+90 555 455 5444

thisiskeithb, 2023-12-28:
> #define STRING_DISTRIBUTION_DATE "2023-10-12"

So you were running this version prior to updating? I think you just happened to get caught in that u8glib error like everyone else and it's not related.

Would you be able to test some [`bugfix-2.1.x` commits](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x) to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works (2023-10-12 in your case). Then, test a commit from halfway between that date and today… And then you keep going to the commit halfway in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

You can also do this manually from the `bugfix-2.1.x` commit history if that's easier:

## Manually Test Commits

1. To manually test commits, browse [`bugfix-2.1.x`'s commit history](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x) using the <kbd>**< >**</kbd> button on the right:
 
   [image]

2. Click on the green <kbd>**< > Code ▼**</kbd> button and then "Download ZIP":

   [image]

3. Apply your minimal config changes, build & flash Marlin to your printer, and observe if the reported issue can be reproduced.

ahmetcemturan, 2023-12-28:
>    the *< >* button on the right:
>    Click on the green *< > Code ▼* button and then "Download ZIP":

Iteration..  OK I will do it..

ahmetcemturan, 2023-12-28:
I fucked up.  Really sorry about it.  When updating the config files I
somehow took the config file of my old board (the fried one) for
reference..  So the board was wrong. It has to be
BOARD_MKS_ROBIN_NANO_V1_3_F4
Noticed that when starting the iteration...
So your fix with the tilde removal worked. (not tested but Printer boots)
Again really sorry.

thisiskeithb, 2023-12-28:
U8Glib version 0.5.4 is now available in PIO's registry, so it'll be used when compiling new firmware (and I opened a PR to update our version to ensure it is used), so this is resolved.

closed 2023-12-28 by thisiskeithb

ahmetcemturan, 2023-12-28:
Y Endstop is unresponsive.  Not inverted. 
 It Works when I disable  //#define FT_MOTION in FW

thinkyhead, 2023-12-29:
> Y Endstop is unresponsive. Not inverted. when I disable `FT_MOTION `

If you can't find a resolution post a new issue about it and there we can work towards a resolution. In the meantime, you might enable `PINS_DEBUGGING` and examine the output of `M43` to make sure nothing is conflicting the Y endstop pin. The `FT_MOTION` feature is pretty new, and it might not be 100% compatible with every type of endstop yet, so we may want to look at that as well.